bigtreetech/BIGTREETECH-TouchScreenFirmware #2456: TFT50 v3.0 No touch since Firmware update

imrdnck1, 2022-03-23:
I updated my Bigtree TFT50 v3.0 to the latest firmware BIGTREE_TFT50_V3.0.27.X and now I have no touch function can any one help?
Thanks.

rondlh, 2022-03-24:
Put a file called "reset.txt" on a sdcard, insert the card into you display and restart the display. This should trigger a touchscreen calibration.

oldman4U, 2022-03-24:
Especially for new users it is important to read the provided information before starting a ticket. Please read the ReadMe, follow the instructions there and close this ticket.

Thank you

imrdnck1, 2022-03-24:
> Put a file called "reset.txt" on a sdcard, insert the card into you display and restart the display. This should trigger a touchscreen calibration.

Thank you.

closed 2022-03-24 by imrdnck1