bigtreetech/BIGTREETECH-TouchScreenFirmware #2457: [BUG] Keep getting popup screen on  TFT70 V3.0

GrooverFromHolland, 2022-03-24:
I keep getting the NO SD CARD popup screen every few seconds. If I acknowledge with OK , it disappears, but comes back immediately or after a few seconds. This makes my touch screen unusable.  Even if I print from USB-drive, this happens.

kisslorand, 2022-03-24:
#2295

GrooverFromHolland, 2022-03-24:
This could be related to https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2295, but it doesn't affect printing.

kisslorand, 2022-03-24:
"NO SD CARD" is a response to M21.

GrooverFromHolland, 2022-03-24:
Yes I know, but what's causing  it. I read all the comments in the relating issue, but that's so much I can't figure out what to do. I have the latest screen firmware and Marlin 2.0.9.3 compiled for BigTreeTech SKR Pro v1.2 (STM32F407ZGT6).

Guilouz, 2022-03-26:
I have similar issue when printing from TFT ports, screen freezing without message and hotend stay on print and keep heating.
No issue when printing from motherboard ports.

This was corrected but since some PR it starts again...

Eddiiie, 2022-03-26:
I am open to it, but how is this a Marlin issue?

Anyone having this issue (I have it too), try moving the ribbon cables around, move power wires away,  see if the messages change.   Just curious.   It seems every time I move the wires around the messages change.

Also, are you using octoprint at all?  Try running in safemode and see if a plug in might be sending something?

kisslorand, 2022-03-26:
> I am open to it, but how is this a Marlin issue?

For example M221 is sent to Marlin and it interprets it as M21.

oldman4U, 2022-04-06:
@Msq001 Please check and close. Thank you

GrooverFromHolland, 2022-04-06:
Why should I close this issue? There is no real solution. Shielding wires and the display makes it less of a problem but does not prevent it.
I's surely a problem of the TFT70 V3.0 display or the BigTreeTech SKR Pro v1.2 board, or the combination of both!
It must be ground loops, or signal integrity. I don’t own the equipment to investigate and solve this myself. I have to rely on the community or BTT engineers to provide a solution.

oldman4U, 2022-04-06:
Because this is not a user group but the software repository related to the firmware made for BTT TFT's. There are user groups out there which can probably help better because they deal with hardware issues more often.

GrooverFromHolland, 2022-04-06:
I don’t think that posting in a user-group will get the attention of the board-designers and engineers to solve this kind of problems. I’m sure software developers and hardware designers work together. Or don’t they?

oldman4U, 2022-04-06:
All developers here are not working for BTT, nor do I. We are all users like you using our spare time to improve this open source project.

GrooverFromHolland, 2022-04-06:
Is it possible to suppress this particular popup in the software? I don’t need it, I print from USB-stick and only need the SD-card reader for flashing the firmware.

digant73, 2022-04-07:
> I have similar issue when printing from TFT ports, screen freezing without message and hotend stay on print and keep heating. No issue when printing from motherboard ports.
> 
> This was corrected but since some PR it starts again...

Do you have this issue with TFT70? if so, did you try with another TFT? is the issue still present?

Guilouz, 2022-04-07:
@digant73 Issue reported in #2480 is fixed with #2479.

But freezing when printing from TFT ports is always present. The screen freezes, no more action is possible on the screen, forced to do a reset and the hotend stops on the current print with the temperatures still active and the motors active.
No issue when printing from motherboard ports (BTT SKR 2).

I don't have other screen to test but it was working fine before with TFT70.

digant73, 2022-04-07:
> @digant73 Issue reported in #2480 is fixed with #2479.
> 
> But freezing when printing from TFT ports is always present. The screen freezes, no more action is possible on the screen, forced to do a reset and the hotend stops on the current print with the temperatures still active and the motors active. No issue when printing from motherboard ports (BTT SKR 2).
> 
> As for the gcode, it's random, on the same gcode the blocking does not necessarily happen in the same place, sometimes after 5min, sometimes 5 hours. Sliced with Cura without any plugin enabled.
> 
> I don't have other screen to test but it was working fine before with TFT70.

I know you reported the progress bar fixed in PR 2470 but (based on the code) it seems to me strange it was fixing that issue. So please provide also the file affected by that issues. I will replicate it and identify the reason of the bug (then probably I will understand how PR 2470 is fixing the bug).
Provide also config.ini and Configuration.h. I don't use Cura and I have no issue. For that reason it is very important to provide the gcode files.
Didn't you have also TFT43, TFT50?

digant73, 2022-04-07:
I would try to set the following settings in config.ini:

language:0
ack_notification:1
prog_disp_type:2
layer_disp_type:0
start_gcode_enabled:0
end_gcode_enabled:0

and verify if there is no freeze.
Reflash the fw (removing the secondary language) and then apply the config.ini with the changes above (use the embedded english language only with no extra language loaded). If there is a bug in the code, I would say it could be related to the above settings. Maybe the freeze is due to french language, ack notification set to 2 or some ACKs sent by Marlin due to M75/M77.

kisslorand, 2022-04-07:
@GrooverFromHolland
> I's surely a problem of the TFT70 V3.0 display or the BigTreeTech SKR Pro v1.2 board, or the combination of both!

As I said earlier it is most probably a Marlin bug. I am in a process of testing a possible fix for it. For 24 hours now I didn't had a single hiccup. 
Which "latest" TFT FW version do you use? The latest "master" or latest "official release"?

Guilouz, 2022-04-07:
@digant73 
prog_disp_type:2 or prog_disp_type:0 (who is the default value) ?

digant73, 2022-04-07:
> @digant73 prog_disp_type:2 or prog_disp_type:0 (who is the default value) ?

I reported to use:

```
prog_disp_type:2
layer_disp_type:0

```
However, I doubt the issue is related to those two settings.

digant73, 2022-04-07:
@GrooverFromHolland Just to identify the root cause of the issue (the SD message provided by Marlin), I would disable the SD CARD (comment SDSUPPORT in Marlin's Configuration.h). Maybe the EMI are affecting the mainboard's SD controller

kisslorand, 2022-04-07:
> 1. Comment SDSUPPORT (in Configuration.h) to disable the SD CARD. You should have no more issues.

In this case he might get `echo:Unknown command: "M21"`...

Guilouz, 2022-04-07:
@digant73 I spoke too quickly, issue for progress bar is still present.

Might be a clue (or nothing to see) but I see that after returning to the menu (after the print summary screen) my printer name (in this case "Super Racer") has changed and shows me "3D printer ready".

gcode I had the problem with: [main droite.gcode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8447333/main.droite.gcode.zip)

[image]

Printing is complete but the progress bar shows 39%

digant73, 2022-04-08:
@Guilouz I see you printed from onboard. In that case the progress is periodically provided by Marlin with M27. It seems that M27 updates have been stopped or are providing wrong progress.

Guilouz, 2022-04-08:
@digant73 What can cause this ? I restarted a print and there the percentage seems correct.

kisslorand, 2022-04-08:
@Guilouz 
You might want to check PR #2491.

digant73, 2022-04-08:
> @digant73 What can cause this ? I restarted a print and there the percentage seems correct.

With the last file (main droite), I just verified that the progress bar was set to 100% while progress value was 6%. I needed to open a menu and come back to Printing menu to display the progress bar at 6%. That is happening because the progress bar is now not supporting any update for a decreasing value although I strongly suggested to not change that because I saw that possibility in the past during some prints and because the progress bar can also be controlled by a remote host with `M118 A1 P0 action:notification Data Left <XXXX>/<YYYY>` and we have no idea how it is used by the remote host. By the way this issue will be fixed simply restoring the previous code

kisslorand, 2022-04-08:
> With the last file (main droite), I just verified that the progress bar was set to 100% while progress value was 6%. I needed to open a menu and come back to Printing menu to display the progress bar at 6%.

This is due to a bug that is fixed in PR #2491, it has nothing to do with not allowing progress bar regression.

digant73, 2022-04-12:
> @GrooverFromHolland Maybe the EMI are affecting the mainboard's SD controller so a false SD insertion was detected, as if you inserted an SD card in the SD slot, and an auto mount was performed by Marlin reporting then the NO SD CARD message to the TFT. Just to validate EMI on mainboard's SD controller, I would try the following two easy solutions in Marlin fw:
> 
> 1. Comment SDSUPPORT (in Configuration.h) to completely disable the onboard SD. You should have no more issues. If so, try the next solution (enable SDSUPPORT and set SD_DETECT_PIN to -1).
> 2. Set SD_DETECT_PIN (in pins_BTT_SKR_PRO_common.h) to -1. With value -1, no auto detect/mount should be performed so you should have no more issues.
> 
> For both the above tests, you must comment the two lines I provided before in parseAck.c (TFT fw). Otherwise the error message (even if still present) will always be suppressed (not displayed) by the TFT.

@GrooverFromHolland Did you have some time for applying the two changes on Marlin reported on my previous post above?
It can help also other people facing the same issue. If you have no time for testing or you are satisfied with your workaround (moving the wires) then please close the ticket.

GrooverFromHolland, 2022-04-12:
Finally testing. I installed the modified screen-firmware with the added lines
{ECHO_NOTIFY_NONE, "echo:No SD card"},         // onboard SD card
{ECHO_NOTIFY_NONE, "echo:SD card"},          // onboard SD card
In in parseAck.c .
Finally, no more popups and still getting SD-card messages in the notification/ info box on the screen, which is GOOD! 
I am not evaluating the other suggestions, because now I can still use the SD-card reader on the screen- board. 
I am not there yet, another popup issue I must deal with now: After Loading filament, not even starting or loading a print, I get “Printer paused, resume printing” that comes back after hitting OK.
Only resetting the screen makes it go away. If I cannot find a solution, I will make this another issue.
Thank You all, I will close this issue in a few days.

digant73, 2022-04-12:
@GrooverFromHolland There is already an open bug report (#2394) for the issue on filament loading. See that.
The two solutions on Marlin fw could fix definitively the issue (and help other users, reporting the same issue) without any workaround on TFT fw (you applied in parseAck.c).

Eddiiie, 2022-04-12:
re Printer Paused,   I no longer get the dialog box up on the TFT on filament change (M600) after a TFT firmware upgrade, but I am able to keep the print going by turning the knob to the right about 1/4 - 1/2 turn then pressing down on the knob.   As if the dialog box was there but not visible and it defaulted to "Pure More" button, turning the knob  (in my head) moves the focus to the other button, "Continue" or whatever it is.   Found it out of frustration.

So, for you, maybe turn the knob and push the button?  ??  Turn other way, push..

closed 2022-04-12 by GrooverFromHolland