bigtreetech/BIGTREETECH-TouchScreenFirmware #2458: [BUG] BLTouch is deployed at startup by TFT HS Mode check command "M401 H"

rondlh, 2022-03-24:
### Description
After the FW update of March 24th I noticed that the nozzle is raised at startup and the bed probe is deployed at TFT startup. 
This should not happen and can cause damage to the probe.

### Steps to reproduce
1. Stow your probe/BLTouch
2. Restart the TFT
3. Watch the probe coming out

**Expected behavior**
Probe should stay stowed, or at least stow again after being deployed.

**Actual behavior**
Probe is deployed and stays that way

### Hardware Variant
BTT TFT35 V3.0

### TFT Firmware Version & Main Board Firmware details
Latest TFT FW (March 25)
MKS SGEN_L V2.0 running the latest stable Marlin 2.0.9.3 (March 18th)

### Additional Information
In parseACK.c I can see the following:
storeCmd("M401 H\n");  // check the state of BLTouch HighSpeed mode

This code is executed during startup. "M401 H" causes my probe/BLTouch to deploy.

kisslorand, 2022-03-24:
#2454
#2452

rondlh, 2022-03-24:
OK, thanks, I close this issue, it's a duplicate

kisslorand, 2022-03-24:
#2459

PR #2459 (2022-03-26): Tiny fixes, tiny program size reduction
### Requirements

BTT or MKS TFT

### Description

- BLTouch HS mode button updates from Marlin's feedback, not from desired value
- tiny program size reduction
- stow BLTouch probe after initial BLTouch HS mode check (guard for older Marlin than 12.III.2022)
- some tiny HU language grammar fix

### Related Issues

Fixes #2458
Fixes #2454
Fixes #2452