bigtreetech/BIGTREETECH-TouchScreenFirmware #2460: [BUG] Strange symbols since recent PRs

Guilouz, 2022-03-25:
Since recent merged PRs, I have strange symbols at startup when screen is not already connected.

TFT70 v3.0

[image]

digant73, 2022-03-25:
probably yes. I also had some freezes with some popups when pressing on the buttons to close the window.
I didn't have that kind of issue before.

Guilouz, 2022-03-25:
Yes there is something wrong with language pack, language file have same name, if I switch to English two languages are named English, if I switch to French two language are names Français.

My french language file is up to date without error.

After switched to English and go back to French strange symbols persist.

[image]
[image]
[image]

Guilouz, 2022-03-25:
I will try your new PR.

Guilouz, 2022-03-25:
Your PR fixes issue with language file name but not for strange symbols.

Guilouz, 2022-03-26:
@guruathwal I printed using the M600 function (not emulated) and during the heating popup in the background we see strange symbols scrolling but also all the parameter strings at the top left.

Msq001, 2022-03-26:
Hello, Please check here https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2466/commits/e58233db1220eb49203da461d4f7207d962b3bf7 in #2466
This problem should be solved

closed 2022-03-26 by Guilouz