bigtreetech/BIGTREETECH-TouchScreenFirmware #2464: [BUG] TFT FW release of March 26th causes TFT lockups

rondlh, 2022-03-26:
### Description
TFT FW release of March 26 causes TFT lockups, usually immediately when I start a print job, I tried starting from SD and USB, different files, started from ESP3D. The result is the same, the TFT freezes.

### Steps to reproduce
1. Start a print job
2. Check if the TFT still responds

**Expected behavior**
Print job should be processed

**Actual behavior**
TFT lockup, motherboard still running.

### Hardware Variant
BTT TFT35 V3.0
MKS SGEN_L V2.0
### TFT Firmware Version & Main Board Firmware details
TFT FW March 26th
Marlin 2.0.9.3 March 18th

### Additional Information
I attached my config.ini for reference.
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8354859/config.zip)

kisslorand, 2022-03-26:
I confirm this bug.

bigtreetech, 2022-03-27:
Please check #2473, it should fix this problem,  Thanks!

rondlh, 2022-03-27:
The issue seems to have been fixed in the update of March 27th, great job!

rondlh, 2022-03-28:
Resolved

closed 2022-03-28 by rondlh