bigtreetech/BIGTREETECH-TouchScreenFirmware #2469: RRF and TFT35_V3.0_E3.27 load\unload filament not work

tribunskiysm, 2022-03-26:
Hi,

SKR 1.4 turbo, RepRapFirmware for LPC176x based Boards 3.3.0_16 (2022-02-13  and TFT35_V3.0_E3 button load\unload filament not work in menu;
No color settings in encoder backlight menu;
The encoder in the print file selection menu does not work; 
in firmware - https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.27.

On the firmware from here - https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update  
encoder, load \ unload filament works.

Help please what am I doing wrong?

radek8, 2022-03-26:
FW TFT has full support for Marlin.
RepRap support is added by some users, but is not 100% implemented.
Do not use old versions of FW. Use a master

tribunskiysm, 2022-03-27:
Latest firmware with RRF support - https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

tribunskiysm, 2022-03-27:
Последняя прошивка с поддержкой RRF - https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

closed 2022-03-27 by tribunskiysm

radek8, 2022-03-27:
Я не заметил, что ты из России.
Я не помогаю оккупантам.

kisslorand, 2022-03-27:
@radek8 
It's not a good idea to bring politics here but I just cannot help myself to let you know I am 100% on the same page as you.

radek8, 2022-03-27:
I'm sorry, but the people of Russia should know that their country is occupying and threatening nuclear war. They won't tell them on TV.
Russia also occupied my country in 1968, and the occupation lasted 20 years.

kisslorand, 2022-03-27:
I know all this and as I said I am with you as most of the world is.
If we open Pandora's box it will be no good. I could write pages of rage every day about what's happening.
It's inhuman, unethical, outrageous, criminal.
BTW, they had in plan to do the same to the country I live in, it was luck it didn't happen.

radek8, 2022-03-27:
ok