bigtreetech/BIGTREETECH-TouchScreenFirmware #2471: [Q] Why am I getting a No Printer Attached?

CaptPickleChips, 2022-03-26:
I am getting a No Printer Attached error on one TFT35 E3 V3, but not another and cannot figure out why.

I have 2x Ender3 Pros that both have a SKR Mini E3 V2, and have been trying to install the TFT35 E3 V3 on the second Ender3 Pro (Blue). The first Ender3 Pro (Red) already has a functioning TFT35 E3 V3. The firmware for both SKR Minis are the same, as well as the firmware for the TFT35. 

I can swap the screens without moving any cables from the main board and the TFT35 from Red will work on Blue, but Blue does not work on Red or Blue. I am at a loss since they are both using the exact same firmware and are installed the exact same way.

rondlh, 2022-03-26:
Check your serial baud rate and serial cable...

CaptPickleChips, 2022-03-26:
> Check your serial baud rate and serial cable...

They are the same. Both printers have the baud rate set to 115200 in Marlin and in the TFT. Again the first TFT I had is recognized by the new printer without any change to the serial cable on the mainboard end, but the new TFT (with all the same firmware and settings) is for some reason not recognized in Touch Mode, despite plugging them in the exact same way.  Marlin Mode works fine for both no problem, so I would assume it has something to do with the serial cable connection, but again, it works on my first TFT.

radek8, 2022-03-26:
I will ask which variant applies.
1. Do both displays work on the first printer and neither on the second printer?
2. Does the first display work on both printers and the second does not work on either printer?

CaptPickleChips, 2022-03-26:
First display works on both printers, second display does not work on either printer.

radek8, 2022-03-27:
There are several possible causes for your problem.
1. Bad configuration. If you have two printers with the same configuration, we can eliminate this problem.
2. Defective cable. Measure the cable or use a cable from a working printer.
3. Cold connection or damaged printed circuit board between processor and pin RX and TX of RS232 connector. Check the TFT board. Be careful not to use any other testers to cause further damage.
4. Defective pin on CPU. It can be remapped to other pins. This has already been addressed several times.

CaptPickleChips, 2022-03-27:
I can eliminate 2. having tried the working cable from printer one. 

3. I am a little new, what exactly would I be looking for on the board?
4. I will look into this, thank you for letting me know.

radek8, 2022-03-28:
Yes. You also have to swap 2 pins on the serial port.

CaptPickleChips, 2022-03-28:
Okay, so that seems to have done the trick! I did not have time to test print yet, but heating up and cooling down functions worked fine! I will update tomorrow!

CaptPickleChips, 2022-03-28:
Starting the printer gives me 2 unknown command errors, M401 and M402, even though I am 100% sure I am using the precompiled firmware for the SKR Mini that does not have the bltouch enabled, any idea what would cause this?

kisslorand, 2022-03-28:
It's a bug. I'll fix it in a moment.

CaptPickleChips, 2022-03-29:
> It's a bug. I'll fix it in a moment. Fixed in PR #2478.

Okay, to be more specific, the first is "M401 H" if that matters.

radek8, 2022-04-03:
@CaptPickleChips  you can close the ticket

CaptPickleChips, 2022-04-03:
My apologies, I meant to a few days ago. Thank you for  all the help, you are amazing.

closed 2022-04-03 by CaptPickleChips

Tomzrk, 2023-03-26:
@radek8 hey man , I've been reading all your comments on TFT issues but couldn't find one similiar to mine , I Recently changed the board on my ender 5 plus and got SKR 3 EZ with TFT35 v3 , I got the No Printer Attached and when i switch to marlin mode i got a black screen with nothing on it , I changed the Serial port on Marlin few times but nothing helped , i connected that screen to a different 3d printer and it was working fine so i swapped the connector between the 2 machines but its still the same , swapped the RX and TX nothing changed so i was wondering if i missed something or its a faulty board

radek8, 2023-03-27:
Hello @Tomzrk 
Many users do not track old closed tickets.
Open your own ticket. Describe your problem. Share your current Marlin configuration, TFT configuration and wiring photos.
Thank you