bigtreetech/BIGTREETECH-TouchScreenFirmware #2475: [BUG] (short description)

dariusz737, 2022-03-27:
### Description

No continous communications with DUET 2 WIFI.

### Steps to reproduce

1. Connect display to DUET 2 WIFI

**Expected behavior**
Continuous two way communication, update between display and board

**Actual behavior**
Single data exchange on power up, or pressing "disconnect" button on display. 

### Hardware Variant
TFT24 - V1.1
### TFT Firmware Version & Main Board Firmware details
TFT 27.x , DUET 3D v 3.4

### Additional Information

digant73, 2022-04-02:
is the baudrate selected for the Wifi port on TFT (Menu->Settings->Connection->S. Ports) the same as the one configured in the Wifi module?

dariusz737, 2022-04-02:
WiFI ? 
Anyway, yes its the same. But there is communication on power up, reset, and when select again „touch” mode. Also warning messages from printer board comes normalny on time.

digant73, 2022-04-02:
so basically you have the issue only in Marlin Mode? If so, you have to set "serial_always_on" to 1 in config.ini or change it from the Feature menu

dariusz737, 2022-04-02:
Załącznik dostępny do 02.05.2022
No, I don’t use Marlin Mode at all. 

Please see attached wideo 
Kliknij, aby pobrać
 <https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FAW_dPSSqKXWCECyJz9J0ytky6jDQAdTpy-IAKCWr3ZpbQj82cAmq9S6U%2F%24%7Bf%7D%3Fo%3DArsRR0L7f3QgK6XiKoNyUlQNh5z74tfThXvOnW7SmcZP%26v%3D1%26x%3D3%26a%3DCAog0BhoOYIailmbo5_WpFkhCsJJBO4DaDkPZ6PwSLvXaZoSdhCJtpvR_i8YicaWpYgwIgEAKgkC6AMA_2pTkchSBDLqMNBaBKr1LpRqJYRUeMeOdXzAQMebG0tzekeBOE66I754_FXOuoKhezj5Raf2z1FyJR5exmQvoCZoanODAOuhe0v83nOXv8NqZraExzDetug6AVOOYDk%26e%3D1651492889%26fl%3D%26r%3D2FB8213D-A260-4F41-B778-B54B7C1167FA-1%26k%3D%24%7Buk%7D%26ckc%3Dcom.apple.largeattachment%26ckz%3D928F0214-9E1C-4125-B372-DCDAF23C4C48%26p%3D105%26s%3DJIkf5pzUAnj8hTDF8QLgvz9mHdc&uk=65lZEhhm22zhljsSXcas-w&f=IMG_3146.MOV&sz=39792747>IMG_3146.MOV
39,8 MB

dariusz737, 2022-04-02:
I have connected display with DUET 2 WIFI board with cable.

digant73, 2022-04-02:
Ok, understood now. It is the first time I see such kind of issue but it seems to me a problem in Marlin. The TFT will connect to mainboard and probably sends the messages but it is like the mainboard is expecting something more before executing the commands received. From the TFT's Terminal menu try to send some commands and verify if you also receive the answer from Marlin (e.g. send M900, M503 etc...)

dariusz737, 2022-04-02:
It's not Marlin, its Duet 3D software.

digant73, 2022-04-02:
ok. simply try to send some commands from TFT's Terminal and verify if you receive the answer (e.g. OK)

dariusz737, 2022-04-02:
Nothing, no response to commands.

digant73, 2022-04-02:
Ok, I was expecting that. You should ask support to duet sw repo. As far as I read on Internet, the Duet sw can be configured to be compatible with Marlin.

oldman4U, 2022-04-03:
Please help the community and close the ticket once you do not need it anymore. Thank you

dariusz737, 2022-04-03:
I need it. Problem still exist. No changes.

oldman4U, 2022-04-03:
But your problem seems not to be BTT TFT related. And this is the repository of the firmware related to this hardware and not a user group. Like digant73 already wrote, maybe someone from the duet repo or a user group related to that hw can help.

dariusz737, 2022-04-04:
I don't understand why are you 100% sure its NOT related do display firmware. But if you say so. Thank you for help.

closed 2022-04-04 by dariusz737

oldman4U, 2022-04-04:
Hi.

I am not sure that it is not the firmware but is there a hint that it is the firmware?

Please reopen it if it is so!

digant73, 2022-04-04:
@dariusz737 see also https://teamgloomy.github.io/tft.html