bigtreetech/BIGTREETECH-TouchScreenFirmware #2480: [BUG] Issue with progress bar and %

Guilouz, 2022-03-28:
Another issue... When print is done, progress bar and percentage are not to 100%. As you can see here they are blocked to 11% on resume page. It's also blocked when printing.

[image]

guruathwal, 2022-03-28:
@Guilouz I have added some changes regarding this. please check and see if #2479 solves it.

kisslorand, 2022-03-28:
@Guilouz 
Does it happen with every print? Can you upload a gcode file?

Guilouz, 2022-03-29:
@kisslorand In test but PR #2479 seem to be fix this.

kisslorand, 2022-03-29:
Even thou I would still like to have the answers because I couldn't reproduce the issue.

kisslorand, 2022-04-08:
@Guilouz 
You might want to reopen this as it is not fixed (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2457#issuecomment-1092289974) and maybe you give more details.

kisslorand, 2022-04-08:
Check PR #2491.

closed 2022-04-20 by bigtreetech