bigtreetech/BIGTREETECH-TouchScreenFirmware #2486: Layer number and progress bar issue

iamacell, 2022-04-03:
Hello All!

SKR1.4T
Marlin 2.0.9.3

BTT TFT35 E3 V3.0
Fw.: V27.x Apr. 02. 2022.

I've enabled all the needed options in Marlin.

I have the following issue:
While printing from Octoprint the progress bar stays empty and no layer numbers shown on the display.
[image]

Is it possible to get the layer number and progress on this screen if printing from octoprint?

digant73, 2022-04-03:
for data progress bar see section "Printing from Remote Host" in README.MD.

iamacell, 2022-04-03:
I've done that already. 
Those make no difference in (not) displaying the layer or progress %, just add a message to the notification area on the top of the screen.

kisslorand, 2022-04-03:
Layer number is displayed only if printing from TFT USB or TFT SD. The info about layer number is taken from the gcode file if the slicer provided that info. Unfortunately Octoprint doesn't send any info about the layer number.

iamacell, 2022-04-03:
I have layer number indicators in my gcodes. Do you mean Octoprint eats these info and not providing to the printer?

kisslorand, 2022-04-03:
Those layer number indicators are marked as comments and yes, Octoprint doesn't analyze comments, it skips them.
The comment parsing is a very new feature of the TFT FW, I do not think anyone else (Octoprint, ESP3D, Marlin, etc.) implemented it yet.

oldman4U, 2022-04-06:
Please help the community and close the ticket once you do not need it anymore. Thank you

digant73, 2022-04-06:
> I've done that already. Those make no difference in (not) displaying the layer or progress %, just add a message to the notification area on the top of the screen.

As reported in README.MD file, the data progress (printing from remote host) is set with:

```
M118 A1 P0 action:notification Data Left <XXXX>/<YYYY>
or
M117 Data Left <XXXX>/<YYYY>

Examples:
M118 A1 P0 action:notification Data Left 123/12345
M117 Data Left 123/12345

```

iamacell, 2022-04-09:
Hello!

I could make the progress bar and elapsed time working with a modified DLP plugin:
[image]

Still no layer number info. Any idea how to tell it to the printer with M118?

kisslorand, 2022-04-09:
It's not implemented in the TFT's FW to parse layer number from anywhere else but as comments from gcode file itself.
However it could be implemented...

iamacell, 2022-04-09:
It would be very nice to have this feature.
Thank you for the info.

iamacell, 2022-04-12:
Thank you for your work!

PR #2496 (2022-04-20): Added Layer Left in Printing from Host API
**IMPROVEMENTS:**
* Added Layer Left to **Printing from Host API** reported in README.MD: It can be used by a remote host (e.g. ESP3D, OctoPrint ect...) to display current print layer number and count. It implements #2486.
* Minor cleanup.

**BUGFIXES:**
* Fixed bug with filament runout on mainboard reported in #2483: After a filament change, the nozzle was moved on top of print and it was needed to press on resume button to resume the print.

resolves #2486
fixes #2483

**PR STATE:** Ready for merge.
closed 2022-04-12 by iamacell