bigtreetech/BIGTREETECH-TouchScreenFirmware #2493: [BUG] TMC driver settings bugs, bump sensitivity and current for Dual X/Y systems

rondlh, 2022-04-09:
### Description

Setting the TMC driver bump sensitivity and stepper current has some issues.

**---Setting bump sensitivity---**
Set X to 1, X2 to 2, Y to 3, Y2 to 4 results in:
X = 2, X2 = 1, Y = 4, Y2 = 3. (Values for X and X2 are swapped, same for Y and Y2)

### Steps to reproduce
1. Set the bump sensitivity for X and X1, or Y and Y1 to different values via the TFT
2. Leave the settings menu
3. Go back into the bump sensitivity menu to see the actually set values.

**Expected behavior**
Values as previously define.

**Actual behavior**
Set value for X and X1 are swapped (same for Y and Y1)

**---Stepper current---**
Setting the current for X will also change the current for X2, the same is true for Y.

### Steps to reproduce
1. Set the TMC driver current for X to a different value than X2 (or use Y and Y2)
2. Leave the settings menu
3. Go back into the driver current menu to check the set value.

**Expected behavior**
X as set, X2 unchanged

**Actual behavior**
X2 is changed to the set value for X. X set correctly.

### Hardware Variant
BTT TFT35 V3.0
MKS MONSTER8 V1.0

### TFT Firmware Version & Main Board Firmware details
Latest TFT FW, April 2nd 2022
Latest Marlin 2.0.9.3 release March 27th 2022

digant73, 2022-04-09:
It seems a bug in Marlin.
If you use pronterface and send M906 Z1052 and then M503 you should see Z2 was also set to 1052.

```
echo:  M906 X1050 Y1150 Z1052
echo:  M906 I1 Z1052

```

rondlh, 2022-04-09:
> It seems a bug in Marlin. If you use pronterface and send M906 Z1052 and then M503 you should see Z2 was also set to 1052.
> echo:  M906 X1050 Y1150 Z1052
> echo:  M906 I1 Z1052
@digant73
Yes, you are right, but the TFT should send "M906 I0 Z1052" when there are more drivers on an axis, then it works as expected.
Did you manage to confirm the bump sensitivity issue?

digant73, 2022-04-09:
No, because I only have X and Y for bump sensitivity. However, you can add `I0` for X, Y etc.. in MachineParameters.c in the following array:

`const char * const parameterCmd[PARAMETERS_COUNT][MAX_ELEMENT_COUNT] = {`

Let me know if it will fix also the issue on bump sensitivity. If so, I suspect I0 must be also provided for other settings (e.g. TMC stealthchop, hybrid threshold etc...)

rondlh, 2022-04-09:
You are right, there is a bug in Marlin, but I also believe the TFT FW has an issue.
M914 X4 should set both X and X2 (if present) according to Marlin.
The TFT **only** sends the M914 commands if something is changed. So if you only change X, then also X2 would be changed if you don't supply the I-parameter.
So your suggestion is excellent, always use the "I" parameter. (MachineParameters.c), using "I0" doesn't cause any issues for Marlin if there is no second driver an the axis.

BTW: It would be nice to also be able to handle the currents, hybrid thresholds and bump sensitivities for Z3 and Z4, but that would make the table bigger and a bit sparse.

digant73, 2022-04-10:
Z3 and Z4 have been added just yesterday by BTT

rondlh, 2022-04-11:
> Z3 and Z4 have been added just yesterday by BTT

Right, I just tested it, and it's working well, and your update even adds the "I0" parameters to the table, great job!

digant73, 2022-04-20:
Ok, so we simply need to wait those fixes are merged in the official Marlin release (not the bugfix) and then we can apply the changes on TFT fw.

rondlh, 2022-04-20:
Yes, kind of, but please note that the old and new meaning of the Marlin I-parameter are incompatible.
Marlin 2.0.9.3 M906, M913:
No I-parameter or I-1 indicates all drivers
I0 indicates the first driver (X)
I1 indicates the 2nd driver (X2)
**That will be updated to:**
No I-parameter or I0 indicates all drivers
I1 indicates the first driver (X)
I2 indicates the 2nd driver (X2)

Marlin 2.0.9.3 M914 is already according to the new convention.

This is bound to cause some issues for some users.

digant73, 2022-04-20:
I understood the logic. But I would wait that both M906, M913 and M914 have the same logic because the piece of code in TFT fw handling those settings is the same. So basically, only M914 is currently not properly working in TFT fw.

rondlh, 2022-04-24:
Because M914 is already converted to the new I-parameter convention in the current Marlin release (2.0.9.3), the I-parameter for X/Y/Z should be "I1", and "I2" for "X2/Y2/Z2", pull request 2493 adds the I-parameter, which is useful for M906 and M913, but still incorrect for M914.

digant73, 2022-04-24:
I know that. I was waiting Marlin was fixed first. However, I just applied the changes to support M914. M906 and M913 will be supported once the same changes already applied to M914 will be applied on Marlin

rondlh, 2022-05-08:
Quick update...
After the Marlin update of May 8th, still version 2.0.9.3, here the status of the TMC commands in relation to the usage of the I-command.

**M913** (Set TMC Hybrid Threshold Speed) and
**M914** (Set TMC stallguard sensitivity)
The meaning of the I-parameter is as follows:
No I-parameter or I0 means all drivers (X, X2)
I1 means driver 1 (X, Y, Z)
I2 means driver 2(X2, Y2, Z2)
The documentation is incorrect for both M913 and M914.

**M906** (set driver current) and
**M569** (Set spreadcycle or stealthchop)
The meaning of the I-parameter is as follows:
No I-parameter or I-1 means all drivers (X, X2)
I0 means driver 1 (X, Y, Z)
I1 means driver 2 (X, Y, Z)

closed 2022-05-09 by bigtreetech

digant73, 2022-05-09:
@rondlh it's better to open a new ticket once the official marlin 2.0.9.4 is available. Just wait for the final solution provided in Marlin before making other changes in TFT fw.

rondlh, 2022-05-09:
I fully agree

digant73, 2022-08-23:
@rondlh Now that Marlin 2.1.1 is released, this topic could be resumed providing the current compatibility matrix.

PR #2503 (2022-05-09): Fixed #2493, #2509, a possible freeze on Status Screen menu
**BUGFIXES:**
* Fixed #2493
* Fixed #2509
* Fixed out of range array size for XYZ status on Status Screen menu: That bug present since #2448 could cause a freeze on Status screen menu. The array size has been restored to 45 (instead of 10).
* Reviewed XYZ status layout for Portrait Mode in Status Screen and Move menus: Z coord was previously not fully displayed (e.g. on TFT35 V3). Now XYZ are properly displayed (see pictures below).

fixes #2493
fixes #2509

**PR STATE:** Ready for merge.

[image]

[image]
commit 8765c1b5 (2022-05-09): Fixed #2493, #2509, a possible freeze on Status Screen menu (#2503)