bigtreetech/BIGTREETECH-TouchScreenFirmware #2501: [FR] - Heater timeout during pause

Help3D-Padova, 2022-04-20:
Filament sensor trigger a Pause and the nozzle go to the park position with the heater on, could be possible to add a configurable timer (inside config.ini) to turn off the heater after a long pause ? There are some materials that, even after a long retraction, suffers degradation due to the heat in the nozzle and sometimes clog it.

@digant73 what do you think ?

digant73, 2022-04-21:
There is a Marlin setting to switch the heaters off after some inactivity period but I do not see any gcode for changing that setting from TFT fw. If existing, it could be simply added in the start gcodes

Help3D-Padova, 2022-04-21:
There is a timer for the stepper motor inactivity but none for the heater, how is called on Marlin ?

digant73, 2022-04-21:
HOTEND_IDLE_TIMEOUT in Configuration_adv.h

Help3D-Padova, 2022-04-21:
There is also 

#define PAUSE_PARK_NOZZLE_TIMEOUT           45  // (seconds) Time limit before the nozzle is turned off for safety.

but is not used because Pause routine is handled by LCD FW right ? 

But using https://marlinfw.org/docs/gcode/M125.html override the LCD pause routine.

digant73, 2022-04-21:
No, PAUSE_PARK_NOZZLE_TIMEOUT is related to M600 and M125 (so it is used only printing from onboard) not from TFT's SD/USB.

Help3D-Padova, 2022-04-21:
My Fault... this was a request for Artillery printers that doesn't have an onboard SD card

digant73, 2022-04-22:
Ok, so you have to use HOTEND_IDLE_TIMEOUT (it's provided exactly for the issue you want to avoid).

Help3D-Padova, 2022-04-22:
Yes ok but the Filament Sensor is not triggering an M600 or M125, i'll try to add it manually in the code :)

digant73, 2022-04-22:
That's correct. The filament sensor will simply trigger a print pause (no gcode is sent). Once in pause (so there is no print activity), the HOTEND_IDLE_TIMEOUT should be used by Marlin after the configured timeout value HOTEND_IDLE_TIMEOUT_SEC.

closed 2022-07-07 by stale[bot]