bigtreetech/BIGTREETECH-TouchScreenFirmware #2506: Homing Failed  printer halted

ssombra, 2022-04-23:
I recently got the BTT TFT35 V3 and I am having some issues.
First I tested it connecting it only with the TFT cable and it works as expected. However, I also want to use it with Marlin mode, so I changed the LCD settings in Marlin as follows:

1. commented my previous LCD (mini12864) and uncommented the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
2. Build Marlin
3. Flash the new firmware to my SKR1.3

The board flashes without problems and I first get a message saying "Failed to enable Bed Leveling" which I understand is normal after an update. Then, when I try to home I get the following message:

Homing Failed:
PRINTER HALTED
Please Reset

I tried going back to my previous LCD and it works again. THE ONLY change I made in Marlin is the LCD.
Any idea?

radek8, 2022-04-24:
Submit your configuration to Marlin

ssombra, 2022-04-24:
> Submit your configuration from Marlin

Sure, both configiguration and configuration_adv?

radek8, 2022-04-24:
Yes, both.
Can you make Home from the touch interface?

radek8, 2022-04-24:
The mistake you have means that there is a problem with some end stop or with one of the stepper motors.
1. Try to move with all the axes
2. Check the end stops with command M119

ssombra, 2022-04-24:
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8550291/Configuration_adv.txt)
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8550292/Configuration.txt)

ssombra, 2022-04-24:
> The mistake you have means that there is a problem with some end stop or with one of the stepper motors.

That's what I would think, but when I change everything back (i.e. the LCD), everything works again.

> 1. Try to move with all the axes
I get the error everytime I want to move something

> 2. Check the end stops with command M119
I am using a BLTouch and this is what M119 shows with the TFT35 V3 setup (the one not working):
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: TRIGGERED
Recv: ok

Which is the same that it shows whit my original setup

radek8, 2022-04-24:
Can you make a video?

ssombra, 2022-04-25:
> Can you make a video?

https://youtu.be/sqfJ95O-T7c

radek8, 2022-04-25:
Changing the display cannot cause this error.

Please send configuration files that are functional with the second display.
Thank you

ssombra, 2022-04-25:
> After a major change in Marlin, you need to initialize the eeprom. Did you do that?
> 
> You have to send two commands after the update. M502 M500

No, I will try that

ssombra, 2022-04-26:
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8559103/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8559104/Configuration_adv.txt)
These are the files that work for me.

ssombra, 2022-04-26:
> After a major change in Marlin, you need to initialize the eeprom. Did you do that?
> 
> You have to send two commands after the update. M502 M500

I tried with M502 and M500 and same error... :(
its frustating...

ssombra, 2022-04-26:
After a lot of tests (and time) adding one section at a time, I finally was able to isolate the problem: 
the EEPROM
IT seems that it got somehow corrupted during my first Marlin flash (after installing the TFT). When I copy all of the settings from my functional conf files BUT the EEPROM settings (define EEPROM_SETTINGS), everything works, including the Marlin mode of the TFT. If I activate the EEPROM settings and flash, I get the error again. For some strange reason, if I flash my original (functional conf files before the change to the TFT35), the EEPROM does work.

I can live without the EEPROM settings but I will continue investigating how to correct that, maybe some sort of EEPROM reset, but I understand this is not a Touchscreen problem but a MArlin one, so I will close this issue. Thanks @radek8 for the support and time.

radek8, 2022-04-26:
You can try to allow (uncomment).
#define EEPROM_AUTO_INIT
It should trigger eeprom initialization on EEPROM error.

Without EEPROM you will not be able to store printer parameters, PID calibration, bed leveling points ....

A physically damaged EEPROM memory area could still be present.
If it is a physical damage to the EEPROM, I would try to emulate the EEPROM on the SD card, or connect an external I2C EEPROM to the I2C connector. This would solve the problem of physical damage and save the chip from further wear and tear.

ssombra, 2022-04-26:
> You can try to allow (uncomment). #define EEPROM_AUTO_INIT It should trigger eeprom initialization on EEPROM error.
>
> Without EEPROM you will not be able to store printer parameters, PID calibration, [...]

I did try that option in Marlin, but it doesn't work. Do you know of a guide on how to emulate the eeprom on the SD card?

closed 2022-04-26 by ssombra

ssombra, 2022-04-26:
> You can try to allow (uncomment). #define EEPROM_AUTO_INIT It should trigger eeprom initialization on EEPROM error.
>
> Without EEPROM you will not be able to store printer parameters, PID calibration, [...]

I tried all possible combination with the EEPROM settings in MArlin:
1)
#define EEPROM_SETTINGS     
#define EEPROM_AUTO_INIT  
#define EEPROM_INIT_NOW  

2)
#define EEPROM_SETTINGS     
//#define EEPROM_AUTO_INIT  
//#define EEPROM_INIT_NOW  

3)
#define EEPROM_SETTINGS     
#define EEPROM_AUTO_INIT  
//#define EEPROM_INIT_NOW  

4)
#define EEPROM_SETTINGS     
//#define EEPROM_AUTO_INIT  
#define EEPROM_INIT_NOW  

and the last one did the trick! it seems to be working now!

radek8, 2022-04-26:
I tried eeprom emulation on SD card on SKR1.4 board
and it worked. I am currently using an external I2C EEPROM.

to file pins_BTT_SKR_V2_0_common.h
insert these lines:

#if NO_EEPROM_SELECTED
     #define SDCARD_EEPROM_EMULATION
#endif

Recompile Marlin.

radek8, 2022-04-26:
You had option 4 at the beginning before you tested.
The EEPROM may be damaged.

ssombra, 2022-04-26:
> You had option 4 at the beginning before you tested. The EEPROM may be damaged.

Yes... it is true. Maybe the process of flashing without any EEPROM setting first and then adding the EEPROM helped... who knows. In any case, if it starts failing again I will consider doing the SD EEPROM emulation. Thanks

radek8, 2022-04-26:
Ok
Good luck.

radek8, 2022-04-26:
I didn't realize you had SKR V1.3
It does not have extra pins on the I2C connector.
But SD card emulation should work

radek8, 2022-04-26:
If you are not using a driver in slot E1, you can also use EEPROM emulation via the I2C connector

ssombra, 2022-04-26:
> Yes... it is true. Maybe the process of flashing without any EEPROM setting first and then adding the EEPROM helped... who knows. In any case, if it starts failing again I will consider doing the SD [...]

> 

Ok, thanks!

radek8, 2022-04-26:
https://www.aliexpress.com/item/1005001632438801.html?algo_pvid=541b971c-815b-423e-852d-1c5bf3e011be&algo_exp_id=541b971c-815b-423e-852d-1c5bf3e011be-7&pdp_ext_f=%7B%22sku_id%22%3A%2212000016899941755%22%7D&pdp_npi=1%40dis%7CCZK%7C%7C11.87%7C%7C%7C%7C%7C%400b0a050b16510127873128954e975a%7C12000016899941755%7Csea&af=3233872_1&cv=34171917&cn=45rayxnbi37iu65zbxszrycnfih45vo6&dp=v5_45rayxnbi37iu65zbxszrycnfih45vo6&af=3233872_1&cv=34171917&cn=45rayxnbi37iu65zbxszrycnfih45vo6&dp=v5_45rayxnbi37iu65zbxszrycnfih45vo6&utm_source=epn&utm_medium=cpa&utm_campaign=3233872_1&utm_content=34171917&product_id=1005001632438801&afref=https%3A%2F%2Falipromo.com%2Fredirect%2Fcpa%2Fo%2Fpqt7yj6tklldo1kil6qc92hevzknxutq%2F&aff_fcid=bee941411e8741df80b905342c2f0c84-1651012823223-04903-_9G57Xi&aff_fsk=_9G57Xi&aff_platform=portals-hotproduct&sk=_9G57Xi&aff_trace_key=bee941411e8741df80b905342c2f0c84-1651012823223-04903-_9G57Xi&terminal_id=290428c8106c48438ea1ffecefca338b&afSmartRedirect=y

ssombra, 2022-04-26:
> 

I am using a second Z driver there...

ssombra, 2022-04-27:
@radek8 after some research in Marlin Github I found a [thread ](https://github.com/MarlinFirmware/Marlin/issues/23436)regarding this bug which seems to be corrected now, but I was still using the latest release. So maybe it is not a EEPROM failure. I will try it tonight.