bigtreetech/BIGTREETECH-TouchScreenFirmware #2507: [BUG] (short description) Not all theme icon filenames are compliant to the naming convention

RedStoneMaster01, 2022-04-25:
### Description

The firmware on the new TFT35 E3 V3.0.1 don't appear to be available. As when I tried to upload the with any of the TFT35 .bmp images for any of the Themes, there was several. There was also issues with "INVALID Keyword's". Now I have a TFT35 that I have to edit .bmp files regardless of theme to get it to work. As well I need to know where/ what file I need to edit to change the 45+ keywords that are not being recognized on the new board revision. 

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

radek8, 2022-04-25:
You may not have used the current files to update.

RedStoneMaster01, 2022-04-25:
Yes, which resolved the "keyword" issue. However, still several .bmp files
were renamed in the update. I had to change several of the .bmp file names
to get them to take.

radek8, 2022-04-25:
Musíš popužít odpovídající verzi souboru BIN, INI a složky s ikonami.

RedStoneMaster01, 2022-04-25:
> Which ones, for example?

TOUCH_MODE was one originally named BIGTREETECH.bmp. Another was USB_DISK,
originally named U_DISK.bmp. Another one was the Delta .bmp. As well were a
few more I had to go in and rename the original .bmp to the the .bin file
to recognize it. After that it all cleared up. This is true for all themes,
I I tested all TFT35 themes before making the ticket. I also want to say
there is another file that is repetitive 6 x's when extracting the zip
file. I don't remember exactly what it was but u think it has to to with
the red theme.

radek8, 2022-04-25:
You can see that you are not using the current files for the update.
The files you report were renamed or changed 3 and 5 months ago.
The correct files are current on github.

radek8, 2022-04-25:
[image]

RedStoneMaster01, 2022-04-25:
No. The bin file I ended up using was updated 16-17 days ago. The time
lapse from the TFT35 E3 V3.0 to the now TFT35 E3 V3.0.1 has been about a
month, roughly.  To get things to work I had to go get the .bin file that
was updated 16 days ago, and the .zip file, as well as the config.ini which
I copied and pasted in the .zip .ini file with custom revision for personal
prefrence, to be able to flash the firmware to the new TFT E3 V3.0.1
correctly.

I would say that is a bit of a run around for new users coming onboard
especially with the new TFT35 E3 V3.0.1 boards hitting the consumer now.

RedStoneMaster01, 2022-04-25:
> Go download the .zip file. Extract it. Those files need updated. It
> honestly don't need all that. Just a simple renaming of the .bmp file will
> allow it to flash correctly.
>
> I have both TFT35 E3 V3.0 AND TFT35 E3 V3.0.1 boards. I can go grab that
> .zip file extract it pull the desired TFT35 folder, .bin file, and .ini
> file and flash to the TFT35 E3 V3.0 display and all works correctly.  To do
> the same with the TFT35 E3 V3.0.1 does not. and that needs noted somewhere
> or the Touchscreen .zip file needs updated.

I'll send the .zip file over to you in a bit. I'm away from the house right
now.

RedStoneMaster01, 2022-04-25:
This is what I'm talking about. Simply downloading, extracting and grabbing
the files needed to flash. Will work with the TFT35 E3 V3.0 however will
not work with the TFT35 E3 V3.0.1 versions of the display. Someone new to
this may be discouraged and mislead when following the process.

As the indication of "latest" is sending a false impression when one has to
run around grabbing updated files from various locations.

For me, after I figured out what was going on. Then is was a simple
changing the .bmp file name, aside from the Delta .bmp which I had to go
fetch as it isn't there at all.

radek8, 2022-04-25:
According to the pin layout, the TFT 35 E3 V3.0 version looks the same as the TFT 35 E3 V3.0.1
Unfortunately, BTT does not specify anywhere TFT 35 E3 V3.0.1.

RedStoneMaster01, 2022-04-25:
Yes, the pin out appears to be the same. According to the date on the box,
the TFT35 E3 V3.0.1 just started being dropped into the market.

Like I said the TFT35 E3 V3.0 date on its box is about 2 weeks earlier if
my memory is correct.  I can check the boxes when I get home.

RedStoneMaster01, 2022-04-25:
And that's what I'm saying a process update needs made cause you can't just
take the source.zip, pull the 1 folder and 2 files need to flash. I just
won't work for the TFT35 E3 V3.0.1. Not without grabbing the other .bin
file, altering the .ini file and change some .bmp file names. That thing
will light up like the 4th of July with read errors.

radek8, 2022-04-25:
What bin file did you use for TFT 35 E3 V3.0.1?

RedStoneMaster01, 2022-04-25:
This bin and ini file work with the TFT35 E3 V3.0.1. The ones in the
source.zip file will throw errors when flashing.

RedStoneMaster01, 2022-04-25:
> <bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2507/1108897111@

Now what I suppose I could do is take those two file's and flash them to
the TFT35 E3 V3.0 and see what happens.

radek8, 2022-04-25:
I do not understand. Which two files?

RedStoneMaster01, 2022-04-25:
Welcome to my world.  This whole mess set me  back about 3 hrs last night.
Because nowhere is there anything indicating the TFT35 E3 V3.0.1.

radek8, 2022-04-25:
I still don't understand.
Maybe someone else will answer you.

digant73, 2022-05-02:
@RedStoneMaster01 Do you realize that you didn't attach any file in this thread (you didn't attach any `source.zip` file)? What was the the fw you were able to flash on the TFT? Was it `BIGTREE_TFT35_V3.0_E3.27.x.bin`? As far as I understood it seems to me that you are still on the original fw installed in the TFT (not the latest fw provided in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update).
Attached you find the latest fw for TFT35 V3.0 E3. Try to install it.

[BIGTREE_TFT35_V3.0_E3.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8601914/BIGTREE_TFT35_V3.0_E3.27.x.zip)

FLYN-001, 2022-05-14:
Hi Im having this issue too, I tried changing the theme on my TFT35 E3 V3 an now whatever theme I try to flash I get various invalid bmp file errors! 
Using the current zip from the master branch. Nothing is working, is there a list of correct & valid file names available?

RedStoneMaster01, 2022-05-14:
What board is the display? V3.0.1?

FLYN-001, 2022-05-14:
> What board is the display? V3.0.1?
> […](#)

Hi, funny thing. I was just gonna come on here to post this when you replied.

I fix my issue by snapping the old slow sd card in half & using a faster micro sd card from the printer in a SD adapter.
The problem was the card was too slow - or was failing.

closed 2022-07-21 by stale[bot]

RedStoneMaster01, 2022-10-11:
Go download the .zip file. Extract it. Those files need updated. It
honestly don't need all that. Just a simple renaming of the .bmp file will
allow it to flash correctly.

I have both TFT35 E3 V3.0 AND TFT35 E3 V3.0.1 boards. I can go grab that
.zip file extract it pull the desired TFT35 folder, .bin file, and .ini
file and flash to the TFT35 E3 V3.0 display and all works correctly.  To do
the same with the TFT35 E3 V3.0.1 does not. and that needs noted somewhere
or the Touchscreen .zip file needs updated.

radek8, 2022-10-11:
HW variant TFT35 E3 V3.0.1 = GD version

ACMadeira, 2023-11-18:
Any update on this?! I've just got a TFT35 E3 V3.0.1 and i couldn't be more regret.. IT never accepts any upgrade, always get's the Fonts: Update Required and Icons: Update Required.. Can't pass throught that screen

ACMadeira, 2023-11-18:
Finally sorted it out.. Add to create files for the Fonts folder (8x16_byte_ascii.fon) And several for the BMP folder(TOUCH_MODE.bmp, USB_DISK.bmp, PRINTING_CHAMBER.bmp, STATUS_CHAMBER.bmp, MPC_PID.bmp and DELTA_CALIBRATE.bmp) For anyone in need of updating the TFT35 E3 V3.0.1 with the a working Vx.x.27 Update, here it goes.
[Backup_Update_TFT 35 E3 V3.0.1.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/13399199/Backup_Update_TFT.35.E3.V3.0.1.zip)

radek8, 2023-11-18:
Sometimes you just need to read the instructions.