bigtreetech/BIGTREETECH-TouchScreenFirmware #2508: Power port on TFT screen

sinole1987, 2022-04-26:
Is the power plug marked in image below a 5v in or out? 
in other words, can I power The screen with a 5v power supply with this port and control the rest of the print with ps_on using the screen?
[image]

rondlh, 2022-04-27:
These are outputs, but they might be suitable for input too, just use a multimeter to check if the +5V pin is directly connected to the +5V off the RS232 port (+ pin in the right side). Please note, the TFT is powered via the RS232 plug connecting the TFT to the motherboard.

oldman4U, 2022-04-28:
Sinole. Please help the community and close the ticket once you do not need it anymore. Thank you

sinole1987, 2022-04-28:
> Sinole. Please help the community and close the ticket once you do not need it anymore. Thank you

I was hoping for more info, I don't really wana mess with board and short it. I already bought 2 of these cause I killed one playing around LOL.

digant73, 2022-04-29:
@sinole1987 +5V pins are all connected to the 3.3V regulator. You can apply 5V on RS232 or on Power connector

sinole1987, 2022-04-29:
> @sinole1987 +5V pins are all connected to the 3.3V regulator. You can apply 5V on RS232, on Power connector, on USB port etc...

so I'm good to go? thank you.

digant73, 2022-04-29:
yes

closed 2022-04-29 by sinole1987

NeoMatrixJR, 2022-05-07:
If these are output...what's the Amperage they support?  Can I use them to power a raspberry pi 4?