bigtreetech/BIGTREETECH-TouchScreenFirmware #2510: font issues with newest firmware on BIGTREETECH TFT35 MZ V3.0

pcclan, 2022-04-28:
been trying to fix this using all the different firmware and nothing is working? is there something i am doing wrong ?i am currently using [BIGTREE_TFT35_V3.0_E3.27.x.bin
[image]

digant73, 2022-04-28:
For MZ version it seems you have to use **BIGTREE_TFT35_V3.0.27.x.bin** (not **BIGTREE_TFT35_V3.0_E3.27.x.bin**).

pcclan, 2022-04-28:
> For MZ version it is possible you have to use **BIGTREE_TFT35_V3.0.27.x.bin** (not **BIGTREE_TFT35_V3.0_E3.27.x.bin**).

i have tried that and it didn't work but if i use BIGTREE_TFT35_V3.0_E3.26.bin the fonts work perfectly

digant73, 2022-05-06:
@pcclan do you have the issue only on that Heat menu or in any menu?

pcclan, 2022-05-06:
> @pcclan do you have the issue only on that Heat menu or in any menu?

any menu that has the centre with temperature or speed of the fans also the terminal with all the keys

digant73, 2022-05-06:
You have to update both the firmware (with BIGTREE_TFT35_V3.0_E3.27.x.bin) and the icon/fonts.

pcclan, 2022-05-06:
> You have to update both the firmware (with BIGTREE_TFT35_V3.0_E3.27.x.bin) and the icon/fonts.

you mean the ones that are in the themes ? because I have tired all the themes

digant73, 2022-05-06:
yes, for example the `TFT35` under folder `THEME_Unified Menu Material theme`. Try to use another SD card

digant73, 2022-05-06:
Try to install this .bin, .ini and icons/fonts

[TFT35_E3.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8639252/TFT35_E3.zip)

pcclan, 2022-05-06:
still the same here are the screens is happening on even with what you sent
[image]
[image]
[image]
[image]

pcclan, 2022-05-07:
fixed when i was able to find a proper sd card just weird it was working with 26 perfectly but didn't like it when it was doing 27

closed 2022-05-07 by pcclan