bigtreetech/BIGTREETECH-TouchScreenFirmware #2511: TFT 7.0 Printer not connected - Line Number is not Last Line Number+1, Last Line: 0

Machinno3D, 2022-04-29:
### Description

After updating the firmware of the TFT 7.0 to be able of using the Smart Sensor the TFT start to buzzing (it doesn't stop after click OK) and gives this message: Printer not connected - Line Number is not Last Line Number+1, Last Line: 0 I'm using the last version of BIGTREE_TFT70_V3.0.27.x and marlin, and, I followed all steps explained. Some times the first time after the update it works, but when I restart the printer the annoying buzzing start again .  It stops when the TFT and the board doesn't have communication.

### Steps to reproduce

1. Download the last version of the firmware (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware)
2. Copy to a SD config.ini, BIGTREE_TFT70_V3.0.27.x.bin and THEME_The Round Miracle Menu Material theme
3. Put the SD in the TFT
4. Update firmware no error messages
5. Change baud rate from 115000 to 250000
6. Error message

**Expected behavior**
Update with no errors and Smart Sensor works well

**Actual behavior**
Printer not connected - Line Number is not Last Line Number+1, Last Line: 0

### Hardware Variant
TFT 7.0
### TFT Firmware Version & Main Board Firmware details
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/63a4832d73cc32aa0f2f1c33852886baf0a0282e/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_GD_TFT24_V1.1.27.x.bin

SKR 2.0 F429 
Marlin-BF-2.0.9.2

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

radek8, 2022-04-30:
> 
> ### Hardware Variant
> TFT 7.0
> 
> ### TFT Firmware Version & Main Board Firmware details
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/63a4832d73cc32aa0f2f1c33852886baf0a0282e/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_GD_TFT24_V1.1.27.x.bin
> 
 But this is not FW for your TFT display!

radek8, 2022-04-30:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2265

radek8, 2022-04-30:
Download the update here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

Machinno3D, 2022-05-02:
Thank you for your prompt reply, I made a mistake copying the link, my bad, but I used your link and now everything is working properly, smart filament sensor and the TFT

Thank you again

closed 2022-05-02 by Machinno3D