bigtreetech/BIGTREETECH-TouchScreenFirmware #2514: M117 displayed while in Touch Mode

ssombra, 2022-05-02:
**Your question or details you need. Please describe.**

I recently changed my old TFT35 V2 with a TFT35 V3.0. After some issues configuring Marlin I finally get the TFT working as expected, both in Marlin and Touch Mode :). 
The only thing I can't have are the M117 messages displayed in the TFT when in Touch Mode. 
In my previous setup, with the TFT35 V2 and an old version of the firmware (sorry, I can't remember the number), I was able to get the M117 messages that I sent from the GCODE displayed in the TFT. I remember that I had to manually change some code following a thread I found here which I can´t find now. 
Now, I configured the TFT35 V3 with the latest firmware (master) and Marlin with also the latest firmware following the instructions in the readme.md page, making sure that all options in Marlin are enabled, particularly those related with dialogs with host, but I can´t have those M117 messages there, the only message that appears is the Printer Ready message. 

**specify the device variant related to your question.**
TFT35 V3.0

digant73, 2022-05-02:
if you are printing from TFT (SD/USB), M177 messages are displayed on Status menu (in infobox area) or in the title bar (in case you are not in the Status menu).
If you are printing from onboard SD (mainboard SD), it is the same as printing from TFT plus you can enable the history (if `notification_m117` is set to `1` in config.ini). If you don't see any notification, the issue is in Marlin side.

ssombra, 2022-05-02:
> if you are printing from TFT (SD/USB), M177 messages are displayed on Status menu (in infobox area) or in the title bar (in case you are not in the Status menu). If you are printing from onboard SD [...]

I think it is definitely a Marlin issue, but I am not sure where to look at. The TFT is connected to my board (SKR 1.3) via its three cables (Exp1, Exp2 and TFT).
 
Regarding communications, I have these settings which are the same I had working on the TFT35 V2:

```
#define SERIAL_PORT       -1 
#define SERIAL_PORT_2     0 
#define BAUDRATE 250000
```

Regarding the dialog with host I have the following:

```
#define HOST_ACTION_COMMANDS 
#if ENABLED(HOST_ACTION_COMMANDS)
  //#define HOST_PAUSE_M76
  #define HOST_PROMPT_SUPPORT 
  //#define HOST_START_MENU_ITEM      // Add a menu item that tells the host to start
  //#define HOST_SHUTDOWN_MENU_ITEM   // Add a menu item that tells the host to shut down
#endif
```

digant73, 2022-05-02:
Try also to enable/set:

EMERGENCY_PARSER (in Configuration_adv.h)
SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)

as reported at the beginning of TFT's config.ini file.

ssombra, 2022-05-02:
> Try also to enable/set:
> 
> EMERGENCY_PARSER (in Configuration_adv.h) SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
> 
> as reported at the beginning of TFT's config.ini file.

They are already set. I enabled everything they suggest but I just wrote in the last post the ones that I believe have to do with communications.

ssombra, 2022-05-03:
Ok, I re-checked my configuration files and, in the Marlin version I am using (bug-fix), I found the following lines after #define HOST_PROMPT_SUPPORT:

  #if ENABLED(HOST_PROMPT_SUPPORT)
    #define HOST_STATUS_NOTIFICATIONS   
  #endif

I enabled #define HOST_STATUS_NOTIFICATIONS and that made the trick. I am not sure if this is something recently added to Marlin, but the readme.md page of this repo doesn't refer to that particular setting. Is it something that should be added to the "Options to support dialog with host" section?

digant73, 2022-05-03:
It is a new setting in Marlin. So when it will be available in the stable Marlin version I will add it in README and config.ini

ssombra, 2022-05-03:
> It is a new setting in Marlin. So when it will be available in the stable Marlin version I will add it in README and config.ini
Thanks for your support, I will close this thread. I hope this info is useful for others.

closed 2022-05-03 by ssombra