bigtreetech/BIGTREETECH-TouchScreenFirmware #2515: [BUG] (TFT70 v3 update fails because of language_x.ini)

davbcr, 2022-05-04:
### Description

FW update fails because of a bunch of errors on labels during language_x.ini update.
List of errors starts with "delta" declarations but also SD card detect.
I compared language_it with language_en and they look the same and i get the very same errors with _en.ini too.

### Steps to reproduce

1. copy content to SD card 
2. insert in TFT card slot
3. switch on and fw update procedure starts

**Expected behavior**
FW to be updated properly

**Actual behavior**
FW update fails

### Hardware Variant
TFT v3.0.27
### TFT Firmware Version & Main Board Firmware details
[Marlin 2.0.9.3 MSU](https://github.com/PierreMasselot1/Marlin/tree/bf95c53990779534b821ca1b06bc9c16b2ad3916)

### Additional Information
[SD card content.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8618361/SD.card.content.zip)

kisslorand, 2022-05-04:
As far as I can see you are uploading a FW from 8th of February 2022 with configuration and language files from 22th of March 2022. I do not know how you ended up mixing it up but it is strongly advised not to do so, keep the configuration and language files together with the firmware they are intended for.

davbcr, 2022-05-04:
I simply downloaded the github repo and try to flash it.
Probably i need to compile it by myself right?

radek8, 2022-05-04:
download files here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

digant73, 2022-05-05:
@davbcr As suggested by radek8, you must download the files from the **master** branch as reported in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#software-repository

radek8, 2022-05-05:
You must use the same version of the file
  * .bin,
config.ini a
language_xx.ini
Why do you use different versions together?

davbcr, 2022-05-06:
I don't use different version intentionally Radek,i fetched updates with GitHub desktop app in First Place and thought It was ok,After i failed First Time i followed your advice downloading the whole master branch with the config.ini alteady had (thinking It was last update) and failed a second time.
Yesterday i downloaded again the whole master branch and tweaked the Just downloaded config.ini but it's not working as well.
The bunch of errors on language_it labels keeps appearing but the rest Is up to date,i mean the files in sd card changed to .cur extension. Just language don't.

digant73, 2022-05-06:
Try to install this .bin and .ini files.

[TFT70.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8638359/TFT70.zip)

Guilouz, 2022-05-06:
Make sure TFT version do you have. GD or not? This happens when you flash a non-GD firmware version on a GD TFT.

davbcr, 2022-05-06:
Sorry to bother but GD stand for what?
btw Bigthreetech TFT70 is a GD one?

Guilouz, 2022-05-06:
the last screens sold now have a GD chipset

davbcr, 2022-05-07:
> Try to install this .bin and .ini files.
> 
> [TFT70.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8638359/TFT70.zip)
> 
> 

I tried,same result : [https://app.box.com/s/v72yi2lhb7n8nlos7f66yyi31eg45uab](video)

davbcr, 2022-05-07:
> the last screens sold now have a GD chipset

Sorry bothering again,so how do i fix It?
I suppose i have to compile custom fw by myself specifying GD chipset in platform.ini file, right?

digant73, 2022-05-07:
The GD versions are also present in this repo https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update. Below the file

[BIGTREE_GD_TFT70_V3.0.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8644790/BIGTREE_GD_TFT70_V3.0.27.x.zip)

davbcr, 2022-05-07:
Thank you. Solved with GD bin file.

closed 2022-05-07 by davbcr