bigtreetech/BIGTREETECH-TouchScreenFirmware #2516: Auto bed leveling / L Corner controls configuration

ssombra, 2022-05-04:
I am still getting used to all the options the firmware has now compared with the old version I was used to use, thanks to the developers for this amazing work. 
I found the "L corner" option under the Auto Bed Level menu 

[image]

and I see that the range for defining the edges is from 60 (min) to 100 (max). 

[image]

Where can I change that range to a much lower minimum?

I searched on the config.ini file and I only find reference to this setting, which was already low, i.e. 10:

```
## Leveling Edge Distance (Manual Leveling)
#   Inset distance from bed edges. This distance is added to minimum X & Y bed coordinates and
#   subtracted from maximum X & Y bed coordinates to calculate manual leveling points.
#   Unit: [distance in mm]
#   Value range: [min: 0, max: 2000]
level_edge_distance:10

```

I also searched in the configuration.h file and I found two references there but none of them was set to the minimum of 60 that I see on my TFT. One under leveling settings:
`#define LEVELING_EDGE_DISTANCE    20  // (mm) Inset distance from bed's edge for calculating leveling point location (Default: 20)
`
and the other under manual leveling:

```
// Manual Leveling
// Move to four corner points to Leveling manually (Point 1, Point 2, Point 3, Point 4).
#define LEVELING_EDGE_DISTANCE_DISPLAY_ID "X/Y"
#define LEVELING_EDGE_DISTANCE_MIN          0                    // Default: 0
#define LEVELING_EDGE_DISTANCE_MAX         100                    // Default: 100

```

SKR 1.3
TFT35 V3.0

digant73, 2022-05-04:
**min** (e.g. 60) = max(probe X, probe Y) + 1 (where probe X and Y are the values defined in `NOZZLE_TO_PROBE_OFFSET` in Marlin's `Configuration.h` or set from Parameter menu in TFT)

**max** (e.g. 100) = LEVELING_EDGE_DISTANCE_MAX (where `LEVELING_EDGE_DISTANCE_MAX` is defined in TFT's `Configuration.h`)

**level_edge** = max(level_edge_distance, LEVELING_EDGE_DISTANCE) (where `level_edge_distance` is defined in TFT's `config.ini` and `LEVELING_EDGE_DISTANCE` is defined in TFT's `Configuration.h`)

ssombra, 2022-05-04:
Thanks, I understand now where the 60 comes from as my NOZZLE_TO_PROBE_OFFSET for X is 59 (59+1=60) and that would be the max of the values. However, that means that the probe uses a very large edge for the corners  (e.g. 60,60 for the left - down corner) because it is assuming that the nozzle can't go beyond the edge of the bed, which in my printer is not the case, am I right?
Is there a way to override that setting so it takes, lets say 10 as the value?

ssombra, 2022-05-04:
I played around with the files and I found in the file "LevelCorner.c" the following setting:

```
uint8_t edge_min = MAX(ABS(getParameter((int16_t)P_PROBE_OFFSET, AXIS_INDEX_X)),
                         ABS((int16_t)getParameter(P_PROBE_OFFSET, AXIS_INDEX_Y))) + 1;
```

If I replace that with:

`uint8_t edge_min = 10;
`

then I have a new minimum value in the Menu that gives me more freedom to choose the level range. I know that my printer can handle a 10mm distance from the bed for probing. Is it safe to do a change like that? i.e. would this affect the firmware in some other undesirably way?

ssombra, 2022-05-04:
Maybe a good alternative would be to use that calculated max value (e.g. 60 in my case) as the default value instead of the minimum value, and let the user change it at will, maybe with a warning of possible collision if not set properly. 
I would love to implement this idea but unfortunately I don't have a programming background.

radek8, 2022-05-04:
@ssombra 
If this area is available for your probe, it's safe.

My probe reaches the entire bed. therefore, I also use the modified LevelCorner.c

[image]

ssombra, 2022-05-04:
> @ssombra If this area is available for your probe, it's safe.
> 
> My probe reaches the entire bed. therefore, I also use the modified LevelCorner.c
> 
> [image]

@radek8 Instead of hardcoding edge_min, what your code does is ignoring the IF comparison first and then using the LEVELING_EDGE_DISTANCE_MIN defined in configuration.h, correct?

radek8, 2022-05-04:
Yes

ssombra, 2022-05-04:
> Yes

Thanks, seems like a better workaround :)

radek8, 2022-05-04:
Your solution is also good

ssombra, 2022-05-05:
@digant73 Thanks! I will try the modifications once they are merged!

Should I close this thread?

digant73, 2022-05-06:
@ssombra no, leave it open to keep track of PR. It will be closed automatically when the PR will be merged

PR #2517 (2022-05-09): Improved Leveling Corner menu
**IMPROVEMENTS:**
* Improved Leveling Corner menu: Edge distance can be set in the range 0-100 (as in Manual Leveling menu). In case the edge distance is below `max(probe offset X, probe offset Y) + 1` then a warning popup message is displayed asking to use the safe min value given by `max(probe offset X, probe offset Y) + 1`. The default edge distance (set when entering in the menu) is given by `max(level_edge_distance, max(probe offset X, probe offset Y) + 1)` where `level_edge_distance` is set in `config.ini`. Furthermore, when exiting from the menu the origina [... cut]
closed 2022-05-09 by bigtreetech