bigtreetech/BIGTREETECH-TouchScreenFirmware #2518: [FR] Mechanical Gantry Calibration Current & Extra Height Adjustment in Touch Mode Under Z Align Menu

sarvenn, 2022-05-07:
It would be nice to have a current and extra height (above z max pos) adjustment in the menu.
Z Align is the perfect place for it. It can be turned to a menu title and sub-menu could have execute auto align, current, extra height titles.

This can be usefull for both single driver dual Z printers. Since I have SKR Mini E3 V2 and a TFT35 E3V3 on my dual Z ender 3 pro, that would be a perfect new feature.
There are already gcode commands for it, so only need is to have buttons and corresponding gcode commands for these buttons.
May be a EPPROM save command also added under Z Align menu.

https://marlinfw.org/docs/gcode/G034-mgc.html

Notes
Requires MECHANICAL_GANTRY_CALIBRATION.

Usage
G34 [S<mA>] [Z<linear>]
Parameters
[S<mA>]	
Current value to use for the raise move. (Default: GANTRY_CALIBRATION_CURRENT)

[Z<linear>]	
Extra distance past Z_MAX_POS to move the Z axis. (Default: GANTRY_CALIBRATION_EXTRA_HEIGHT)

I also requested Marlin UI version for this feature request. Touch mode can be implemented without any update need from Marlin side.
https://github.com/MarlinFirmware/Marlin/issues/24145

sarvenn, 2022-07-07:
@kisslorand @radek8 @guruathwal @digant73 @Msq001 
Hi,
Could anyone help on this new feature request please?
I think it is quite straight forward. This would with perfectly under Z Align Menu for changing the parameters for next Z align command, and we also need a eeprom save button nearby :)

sarvenn, 2022-08-16:
@kisslorand @radek8 @guruathwal @digant73 @Msq001
would you please give a chance to my FR?

kisslorand, 2022-08-16:
Sorry, I do not have the necessary HW to test anything I would do in regard to this FR.

radek8, 2022-08-16:
I also don't have HW that supports the G34 command

digant73, 2022-08-17:
This PR cannot be currently implemented for the following reasons:
1) Z_STEPPER_AUTO_ALIGN and MECHANICAL_GANTRY_CALIBRATION are mutually exclusive in Marlin and for the TFT it will be not possible to choose the correct G34 menu to show (G34 for auto align not requiring any param or G34 for mechanical gantry requiring two param?).
2) It requires to provide dangerous specific mainboard info such as current for which the TFT should be able to retrieve and display the default values configured in Marlin (e.g. GANTRY_CALIBRATION_CURRENT). But even in this case, the TFT has no possibility to retrieve those default values.
With these limitations, a TFT user having no clear idea of that functionality could easily provide a completely wrong value damaging the mainboard's components.

sarvenn, 2022-08-17:
I have an Ender 3 Pro with SKR Mini E3 v2 (single Z step driver with 2 Z motor cable slots) and a TFT35 E3 V3.
I think there is no need to have double Z motors to test it.
But Antotino stated a different point that I could not consider before.
So, if it is technically has a potential to damage a board due to different Z stepper (single or dual) applications and we are sure about that, then I may close my FR.
As I up to know, dual Z stepper application uses BLTouch to align the X gantry, but with single stepper driver it goes to Z max point and touches to a rigid place, reduces the stepper current and still drives the motor to align the gantry. Tunable parameters are for the current and z distance for aligning the gantry.

digant73, 2022-08-17:
it's not a problem due to different Z stepper (single or dual) but due to current Marlin implementation. Marlin should provide information about the provided G34 implementation (e.g. a capability reporting G34 for Z auto align or Mechanical gantry). For Mech gantry it should also provide the default values for current and height so the new menu on TFT could display them as initial values. Once you change the values sending G34 S<new current> Z<new height>, Marlin should also be able to provide the new values currently set (e.g. in case G34 without any param is sent by the TFT, Marlin should reply with an ACK reporting the current values). But current Marlin fw is not providing anything of that needed by to properly implement new menu on TFT.
Z Align does not need any "mandatory" param so there was no problem to implement a menu on TFT. If you don't need to specify any param for the Mech gantry version of G34 you can use the existing menu on TFT. Obviously instead of Z align a mech gantry (with default values) will be executed by Marlin.

sarvenn, 2022-08-17:
Thanks for the detailed explanation Antonino.
Maybe I can raise a FR at Marlin side at first.

closed 2022-08-17 by sarvenn

digant73, 2022-08-17:
Yes, you should ask on Marlin side to provide such kind of info in order TFT can properly handle G34 menus

kisslorand, 2022-08-18:
While at it one can always implement this for own usage by using the custom gcode menu.

sarvenn, 2022-08-18:
> While at it one can always implement this for own usage by using the custom gcode menu.

Yeah, for sure, I'll give it a try.
Actually these parameters do not frequently change.