bigtreetech/BIGTREETECH-TouchScreenFirmware #2527: [Q] (Rotary button LED)

BelSD, 2022-05-26:
I just upgraded from version v26 to v27 and the rotary button LED is no longer lit and in the menus I no longer have the possibility of activating it or even changing its color. The options in the menu have disappeared. Is this normal or a BUG?

[image]
[image]
[image]

radek8, 2022-05-26:
You used the wrong update file

oldman4U, 2022-05-27:
BelSD

Please answer or close this ticket.

Thank you

BelSD, 2022-05-27:
What wrong update file?
I downloaded the source code (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware) and compiled

radek8, 2022-05-27:
If you compiled your own FW, then you compiled the wrong versions.
According to the photos, your display is a variant of the TFT35 E3 V3.
You have compiled versions of TFT35 V3.0

The difference is that the variant without the E3 index does not have an illuminated knob.

radek8, 2022-05-27:
Variant E3 has a knob at the bottom.
The version without E3 has a knob on the side

BelSD, 2022-05-27:
Thanks for Help

closed 2022-05-27 by BelSD