bigtreetech/BIGTREETECH-TouchScreenFirmware #2530: [BUG] BTT GD TFT35 and SKR mini e3 v3 on Ender3 v2 freezes while printing

Sw1jge, 2022-05-29:
This happens when I try to print through the usb that screen have.
Freezing happens completely randomly at random times. At the same print gcode at 20 hours total printing time the first time freezing about 5 hours after its start and the second time freezing 20 minutes before finishing the print.
After the freeze the hotend and the bed maintain print temperatures and I can not turn them off unless I do restart to printer .
Yesterday I make an update to the last version of my screen and I hope this fix this error.
Any idea whats going on here?
Thanx and sorry for my poor english.

Guilouz, 2022-05-31:
There is similar issue with SKR2 and MKS robin Nano V3 boards. Issue only appears when printing from SD/USB ports of the TFT, not from motherboard ports. I use TFT70 no GD.

Hotend stopped on the print, hotend and the bed maintain print temperatures, the motors remain active and the screen no longer responds and freeze, having to turn the printer off and on again and at the same time losing the print in progress.

I have tried several settings in config.ini as @digant73 suggested to me but without success.

closed 2022-05-31 by Sw1jge

Sw1jge, 2022-05-31:
Thanx Guilouz
I have search to many topics about that issue and I cant find any solution to this problem.
You have absolutly right to all. If i printing from motherboard usb or with octoprint none printing its freeze.
After the last update i have done in my screen I made two prints about 4 hours the first and 5:30 hours the second and fortunately it didnt freeze.
Fix the issue? I dont know but I hope so . 
If I want to print something i prefer to doing with motherboard usb and octoprint.