bigtreetech/BIGTREETECH-TouchScreenFirmware #2536: I flashed the latest version of BIGTREE_TFT35_V3.0.27.x and homing fails every time

Mech131, 2022-06-09:
### Description

After flashing BIGTREE_TFT35_V3.0.27.x, BLTouch no longer works. It was working on the latest marlin bug_fix release, 2.1 and previous version just fine.

### Steps to reproduce

1. Flash latest BIGTREE_TFT35_V3.0.27.x on Ender 5 Plus
2. Do anything to enable the BLTouch
3. Machine Hault

**Expected behavior**
Print to start, be able to set P-Offset, etc...

**Actual behavior**
HAULT

### Hardware Variant
BIGTREE_TFT35_V3.0.27.x w/ BTT E3V3
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0.27.x w/ BTT E3V3

### Additional Information
[Config.h](https://drive.google.com/file/d/1N-MVw9wAHJtopJTAKAT2GnHfjzafnn0I/view?usp=sharing)
[Config_adv.h](https://drive.google.com/file/d/1aPTPmQqIus7HdWT5kIZ9ZSKqQEGD7nqP/view?usp=sharing)

Mech131, 2022-06-10:
When i first experience the failure, it was through activating it with Octoprint. From there on, I tried activating it from the TFT35 screen and the issue persists. Are old versions of the TFT screen firmware available for download? I would love to flash back to my old version and get my device up and running again.

kisslorand, 2022-06-10:
What was Octoprint connected to? TFT? Motherboard?

Mech131, 2022-06-10:
Motherboard

kisslorand, 2022-06-11:
In this case this issue is not related to the TFT's FW. The TFT cannot interfere to a command issued to the MB from an external source connected to the motherboard.

Mech131, 2022-06-11:
> In this case this issue is not related to the TFT's FW. The TFT cannot interfere to a command issued to the MB from an external source connected to the motherboard.

To test that I removed the TFT from the BTT E3V3 and the BLTouch started working again. When i reconnect the TFT to the BTT, the BLTouch stops working again.

kisslorand, 2022-06-12:
What if you flash an older version (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases)?

Mech131, 2022-06-14:
> What if you flash an older version (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases)?

Flashed [Vx.x.23](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.23) and it works fine.

kisslorand, 2022-06-14:
What file/environment did you select? There is no BTT E3V3 in Vx.x.23.

Mech131, 2022-06-14:
My apologies, it was version [Vx.x.25](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.25). I used "BIGTREE_TFT35_V3.0.25.bin"

kisslorand, 2022-06-14:
OK. What file did you use from Vx.x.27? You should use "BIGTREE_TFT35_V3.0_E3.27.x.bin", not "BIGTREE_TFT35_V3.0.27.x" as you stated in OP.

oldman4U, 2022-06-14:
He read the readme and used the correct firmware. Please see the reply from 5 days ago.

radek8, 2022-06-14:
The only difference between the TFT35 V3 and TFT35 V3E3 is in the knob backlight. This will not do the problem.

kisslorand, 2022-06-14:
OP stated that with older FW the BLTouch works. I do not remember any changes in TFT FW that could result in higher power consumption.

radek8, 2022-06-14:
I just thought out loud .... When the brightness changes, the current also changes

oldman4U, 2022-06-18:
Mech131 what is the status of this ticket?

Mech131, 2022-06-19:
It was resolved by installing the original firmware that came on my device, version [Vx.x.25](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.25)

oldman4U, 2022-06-20:
OK

digant73, 2022-08-14:
@Mech131

First, you have to enable the following (missing) settings in Marlin (as reported on top of config,ini file):
`BABYSTEPPING`
`M115_GEOMETRY_REPORT`
`LONG_FILENAME_HOST_SUPPORT`
`AUTO_REPORT_SD_STATUS`

Then, try to disable:
ADVANCED_OK

Then, if you have a genuine (or clone) of BLTouch 3.0/3.1 and a 5V-tolerant controller on mainboard, try to enable:
BLTOUCH_SET_5V_MODE

radek8, 2022-08-14:
I would also allow
#define BLTOUCH_DELAY 500
Some BL-TOUCH have a longer processing time and can cause problems.

Mech131, 2022-08-18:
@digant73 What does M115_GEOMETRY_REPORT (in Configuration_adv.h) do?

Mech131, 2022-08-21:
Updating to the latest version of the TFT35 firmware that was released in October/2022 OR downgrading to the firmware it shipped with resolved the issue of the BL-Touch not working at all. Now the problem is intermittent probing failure that seems related to the board, not the TFT35 screen.

closed 2022-08-21 by Mech131

digant73, 2022-08-22:
Probably you meant latest TFT fw released in August 2022 (not October).
Anyhow, thanks for the clarification.