bigtreetech/BIGTREETECH-TouchScreenFirmware #2537: ABL Buttons Missing with RRF

Wombat37, 2022-06-09:
After updating to recent firmware on my E3 RRF/TFT35 E3 V3 system, I seem to have lost the ABL buttons on the TFT. Has something changed so that these buttons are now disabled?

Wombat37, 2022-06-14:
OK, I'm not seeing any feedback - perhaps I need to provide more details.

Following latest update, TFT displays firmware version as V27.x April 9 2022

Movement screen now appears as follows:

[image]

Note the missing Bed Level icon

I installed a later firmware version V27.x May 30 2022 and still the Bed Level icon was missing

I reinstalled the earlier version TFT35_E3_V3.0.27x Sep 13 2021 and the Bed Level icon reappeared as seen below:

[image]

This gives me access to the various ABL functions:

[image]

On each occasion, I'm updating the BIN file, the renamed config_rrf.ini file and the TFT35 folder. 

All I'm doing is just changing the firmware and ABL appears with older versions but not the newer versions. No other changes are being made. Hence my original question. Am I missing something here, is this a bug or a deliberate change?

Thanks for any guidance anyone can give me.

kisslorand, 2022-06-14:
Is you BLTouch enabled?

radek8, 2022-06-14:
@Wombat37 
September 13 icon is fine, May 30 is missing.
This is a very long time.
Can you try FW from another date in this section to find out when it broke?
Thank you.

Wombat37, 2022-06-14:
Thank you for your help.

I'll try and answer your questions:

1. Is BL Touch Enabled?
My understanding is that the TFT firmware will automatically detect the BLTouch - presumably from the RRF configuration.
I have configured a Type 9 probe and ABL in my RRF config.g file:

; Z-Probe  Bltouch
M950 S0 C"servo0"                               ; create servo pin 0 for BLTouch
M558 P9 C"^probe" H5 F120 T6000    ; set Z probe type to bltouch and the dive height + speeds
G31 P500 X-44.36 Y-4.27 Z0.800          ; set Z probe trigger value, offset and trigger height 

;Define ABL mesh grid
M557 X20:200 Y20:220 P4:4		;Use number of points**

2. Long time between updates
I was very happy with the Sep 13 2021 version and so I felt that if it's working, don't fix it. However I kept getting into a mess with the z-offset after using baby-stepping that when I saw the bug fix for this, I updated. The z-offset problems went away but so did the bed leveling control :)

3. Try interim versions
I will do that - probably will start with @radek8's suggestion of the Sep 22 release. I'll post results shortly.

4. RRF M115 response

M115
FIRMWARE_NAME: RepRapFirmware for STM32F4 based Boards FIRMWARE_VERSION: 3.3.0_9 ELECTRONICS: STM32F4 FIRMWARE_DATE: 2021-09-19 11:51:46

5. M503 Response

M503
; Configuration file for BTT-E3-RRF (firmware version 3)
; executed by the firmware on start-up
;
; generated by RepRapFirmware Configuration Tool v3.1.10+4-LPC on Tue Jan 19 2021 17:25:01 GMT+0800 (中国标准时间)

; General preferences
G90                                            ; send absolute coordinates...
M83                                            ; ...but relative extruder moves
;M550 P"BTT-E3-RRF"                         ; set printer name
M550 P"andys3dprinter"                         ; set printer name

; Network
M552 S0					;Comment-out for FW update

M586 P0 S1                                     ; enable HTTP
M586 P1 S0                                     ; disable FTP
M586 P2 S0                                     ; disable Telnet

; Drives
M569 P0 S0                                     ; physical drive 0 goes forwards using default driver timings
M569 P1 S0                                     ; physical drive 1 goes forwards using default driver timings
M569 P2 S1                                     ; physical drive 2 goes forwards using default driver timings
M569 P3 S0                                     ; physical drive 3 goes forwards using default driver timings
M584 X0 Y1 Z2 E3                               ; set drive mapping
M350 X16 Y16 Z16 E16 I1                        ; configure microstepping with interpolation
M92 X80.00 Y80.00 Z400.00 E406.86               ; set steps per mm
M566 X900.00 Y900.00 Z60.00 E120.00            ; set maximum instantaneous speed changes (mm/min)
M203 X12000.00 Y12000.00 Z600.00 E3000.00     ; set maximum speeds (mm/min)
M201 X500.00 Y500.00 Z100.00 E5000.00            ; set accelerations (mm/s^2)
M906 X580 Y580 Z1000 E650 I30                   ; set motor currents (mA) and motor idle factor in per cent
M84 S30                                        ; Set idle timeout

; Axis Limits
M208 X0 Y-6 Z-1 S1                               ; set axis minima
;M208 X230 Y230 Z250 S0                         ; set axis maxima
M208 X250 Y250 Z250 S0                         ; set axis maxima X set to 250 to allow 3D Touch to reach further to the right

; Endstops
M574 X1 S1 P"xstop"                            ; configure active-high endstop for low end on X via pin xstop
M574 Y1 S1 P"ystop"                            ; configure active-high endstop for low end on Y via pin ystop
[... 26 lines cut ...]
; Fans
M950 F0 C"fan0" Q1000                          ; create fan 0 on pin fan0 and set its frequency
M106 P0 S0 H-1                                 ; set fan 0 value. Thermostatic control is turned off
M950 F1 C"fan1" Q1000                          ; create fan 1 on pin fan1 and set its frequency
M106 P1 S0 H1 T60                              ; set fan 1 value. Thermostatic control is turned on

; Tools
M563 P0 D0 H1 F0                               ; define tool 0
G10 P0 X0 Y0 Z0                                ; set tool 0 axis offsets
G10 P0 R0 S0                                   ; set initial tool 0 active and standby temperatures to 0C

;Andy's workaround for M221 No Tool Selected error
M104 S0 T0
T0

; Custom settings are not defined

; Miscellaneous
;M575 P1 S1 B57600                              ; enable support for PanelDue
M575 P1 S2 B57600

M552 S1                                        ; enable network. Comment-out for F/W update
M150 X2

M501											;Call config-override.g (for PID tuning etc.)

Wombat37, 2022-06-14:
Further Testing (this was very tedious)...

**Version Merged/Bed Level Icon**
Sep 22 2021 (#2179)/Yes
Dec 30 2021 (#2307)/No
Nov 9 2021 (#2233)/No
Oct 22 2021 (#2222)/No
Sep 21 2021 (#2194)/Yes
Sep 23 2021 (#2195)/404 Error
Sep 27 2021 (#2199)/Files deleted
Sep 27 2021 (#2201)/404 Error
Oct 5 2021 (#2209)/404 Error

There seemed to be a lot of activity in developing code around the end of September 2021 - much appeared to be in parallel and the development and the merged dates didn't seem to keep in order.

The Bed Level icon was present in the Sep 21 and 22 versions and then all the following files were missing from the repository until Oct 22 when the icon disappeared - so it appears that the problem was probably introduced in one of these missing versions that I couldn't test.

If someone could get these missing (compiled) files to me, I will test them.

Thanks again for your support.

Wombat37, 2022-06-14:
My money's on the Oct 5 2021 version - there's a lot of code modification around ABL in there.

Wombat37, 2022-06-14:
Accidentally closed this :(

radek8, 2022-06-14:
Try this version.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/aecf5b3f513f64932d525c62dbc3ec5d66a9d61d/Copy%20to%20SD%20Card%20root%20directory%20to%20update

Wombat37, 2022-06-14:
I tried the version you sent - there is NO Bed Level icon in this version. So the issue may lay in here. There a persistent TFT in Listening Mode message.

I understand that this code is maintained mainly by volunteers and I'm very grateful for all their efforts.

This is a function that was working previously and so I'm hopeful that this issue can be an easy fix.

kisslorand, 2022-06-14:
@Wombat37 
Set "BED_LEVELING_TYPE" in "Configuration.h" to the value that suits your setup. Check if it solves your problem.

radek8, 2022-06-14:
Yes, it's from the latest branch.
If it works for you, you can use it.
I will publish the correction.

Wombat37, 2022-06-14:
Thanks again!

I'm surprised no one else noticed this :)

Shall I close this out or do we wait for formal release?

radek8, 2022-06-14:
OK, I was happy to help.
The ticket is closed by merging the repair.

kisslorand, 2022-06-14:
@radek
PR #2541 has the same effect as my proposal about setting "BED_LEVELING_TYPE" in "Configuration.h".
On another note, do all Reprap FW printers have ABL???

radek8, 2022-06-14:
@kisslorand 
I'm afraid your solution with Configuration.h won't work.
parameter #define AUTO_LOAD_LEVELING
is set to 1 by default
Therefore, precompiled versions have leveling enabled in Configuration.h. Still, it didn't work for the RRF.
Configuration.h sets automatic detection, but if leveling is not detected, the default value is  leveling disabled.
Currently, TFT cannot automatically detect leveling and its type at RRF, so I set the leveling ABL to RRF fixed.
If someone else contacts us using another method, we can define it, for example, in the Config .ini file.
Or someone who knows the RRF better and writes a better correction.

radek8, 2022-06-14:
I asked Wombat37 for an M503 and M115 listing
Unfortunately, I did not find any string according to which it is possible to find out if leveling is active and what method it uses.

radek8, 2022-06-14:
@Wombat37
What leveling methods can be defined in RRF?

kisslorand, 2022-06-14:
> I asked Wombat37 for an M503 and M115 listing Unfortunately, I did not find any string according to which it is possible to find out if leveling is active and what method it uses.

This is why it should be user configurable, not baked in the TFT's FW.

radek8, 2022-06-14:
However, no other options are defined in the comment.
The user cannot know that there are other options.

[image]

[image]

kisslorand, 2022-06-14:
This has nothing to do with the issue. This parameter is effective ONLY if the printer has ABL.

radek8, 2022-06-14:
Configuration.h sets
infoSettings.auto_load_leveling on 1.
But is defined in the setings.c file:
infoMachineSettings.leveling = BL_DISABLED;
And if autodetection does not overwrite it, the leveling will remain deactivated.
Therefore, the settings in Configuration.h do not resolve the issue.

kisslorand, 2022-06-14:
I beg to differ. Check setupMachine() in settings.c.

radek8, 2022-06-14:
ok

radek8, 2022-06-14:
What about this?

ABL is set during automatic detection and there is an option to select manually.

[image]

Wombat37, 2022-06-14:
In this comment, I try to describe how RRF handles ABL.

It looks like you just need 4 entries in the RRF config.g macro - which is run each time the system boots.

M950 S0 C"servo0"                                      ; create servo pin 0 for BLTouch

M558 P9 C"^probe" H5 F120 T6000           ; set Z probe type to bltouch and the dive height + speeds
;The P9 selects the BLTouch type of probe - other options can be selected here

G31 P500 X-44.36 Y-4.27 Z0.800                 ; set Z probe trigger value, offset and trigger height 

M557 X20:200 Y20:220 P4:4			    ;Set the mesh test points

There is no option to select the leveling 'mode' along the following lines in the TFT Configuration.h file

 * Bed Leveling Type
 * The TFT will attempt to auto detect bed leveling type with marlin firmware.
 * Set to 1 to enable auto-detect (Marlin only) or choose one of the options
 * below if auto-detect does not work.
 *
 * WARNING: If you're not sure, leave on auto-detect or disabled.
 *          UBL has extra options other leveling systems might not have.
 *
 *   Options: [disable: 0, auto-detect: 1, ABL: 2, BBL: 3, UBL: 4, MBL: 5]
 */
#define BED_LEVELING_TYPE 1  // Default: 1

I don't even know what these other options mean (BBL, UBL and MBL) and they don't seem to be present in the RRF config.g.

Once, the entries in the config.g file have been made, the ABL can be invoked from the DWC webpage as follows:

[image]

G29 runs the ABL matrix directly whereas G32 calls it indirectly via a macro (so that other commands like positioning and messaging can be added).

If the BLTouch is disabled (M558 P0), the same ABL options are still presented. In other words ABL access is not dependent on a suitable probe being present. I think what it does is take the user through a set a manual adjustments to get the ABL. I couldn't test this because I needed to home the nozzle - something I couldn't do once I'd disabled the BLTouch.

So, I think what RRF does is always make ABL available and use BLTouch (or any other type of sensor) if present, otherwise take the user through a series of manual settings. In other words, RRF always has ABL enabled.

I hope this helps.

P.S. How did it work in earlier versions of the firmware?

Wombat37, 2022-06-15:
OK - I looked into the Marlin leveling modes ABL (basic linear), UBL (unified ), BBL (bilinear) and MBL (manual, I assume).

These are different ways of calculating the mesh compensation from assuming a flat plate at an angle to interpolations between measured nodes.  

In RRF, there seems to be only one ABL option and it looks like a sophisticated curve-fit between the test nodes - so it's already equal or better than all the Marlin modes so a choice it not required (except manual which is automatically applied in the absence of a z-probe).

So, again, I think we can get away with enabling ABL by default on RRF systems.

It's difficult to find information on the mesh fitting algorithms on the two systems and I'm making a few assumptions here. It would be good if we reached out to the RRF community to confirm things. I can do that.

kisslorand, 2022-06-15:
> P.S. How did it work in earlier versions of the firmware?

Before PR #2179 a few things were set by default if the detected firmware wasn't Marlin. Amongst them was the enabling of leveling option too.
```
  if (infoMachineSettings.firmwareType != FW_MARLIN)  // Smoothieware does not report detailed M115 capabilities
  {
    infoMachineSettings.EEPROM                  = ENABLED;
    infoMachineSettings.autoReportTemp          = DISABLED;
    infoMachineSettings.autoReportPos           = DISABLED;
    infoMachineSettings.leveling                = ENABLED;
    infoMachineSettings.zProbe                  = ENABLED;
    infoMachineSettings.levelingData            = ENABLED;
    infoMachineSettings.emergencyParser         = ENABLED;
    infoMachineSettings.autoReportSDStatus      = DISABLED;
  }
```

I am not familiar with RRF and I do not know if those settings that were eliminated are needed or not to be set by default for RRF.

kisslorand, 2022-06-15:
> What about this?
> 
> ABL is set during automatic detection and there is an option to select manually.
> 
> [image]

If RRF has only one type of bed leveling and it always have this option than I think it is just better to leave it as you did originally. Even if the user sets any other type than ABL, RRF doesn't have that so defaulting to ABL is not wrong.
I assume all these based on the info  Wombat37 gave us.

Wombat37, 2022-06-16:
OK, I did the following tests with the build @radek8 sent me on my E3 RRF/TFT35 E3 3.0:

**With BLTouch configured with M558 and G31 and ABL enabled with M557**

- All ABL and BLTouch functions seemed to work. Can't test bed leveling as I have only single z-axis motor.

**With BLTouch configured but no M557**

- All BLTouch functions accessible and work
- ABL nothing happens. This is not causing any damage but RRF/DWC gives the error message below, TFT does not give any message. 
- [image]

**No BLTouch but ABL enabled with M557** 

- Had to reconfigure z-axis microswitch to home
- BLTouch buttons do nothing. No error messages given.
- ABL on the TFT gives the following message and then mainboard crashes and resets
[image]
- ABL on DWC does the same thing - same message and subsequent mainboard crash and reset
[image]

Summary - with ABL always enabled, the TFT behaves essentially the same as the RRF/DWC. The only difference is that in the absence of a M557 in the RRF config.g file, both systems do nothing but DWC gives an error message and the TFT does not. I don't think this is a critical difference.
I thought that with the BLTouch disabled and the ABL enabled with M557, a manual ABL process would start - it didn't. Both DWC and TFT gave error messages and then crashed and reset the mainboard. So the biggest issue is pressing the ABL button on the TFT without the M557 statement present - it gives a system crash. RRF/DWC does exactly the same.

My recommendation is that we go with this build @radek8 sent me to test. Although not perfect, no damage is caused and it works when stuff is configured properly. It would be good if we could capture this stuff in the user manual.

radek8, 2022-06-16:
Should the display of the leveling button be conditioned by the presence of parameters M557?

radek8, 2022-06-16:
@Wombat37
you can try the enclosed FW
it should detect the presence of M557 and enable leveling icons accordingly.
If the M557 command is missing from the configuration, the leveling icons will not be displayed.
[BIGTREE_TFT35_V3.0_E3.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8921946/BIGTREE_TFT35_V3.0_E3.27.x.zip)

Wombat37, 2022-06-16:
I did some further testing with @radek8's previous build. I found that the mesh grid definition (M557) **must** be preceded by a definition of the z-probe type (M558).  I assumed that if I just removed the M558 instruction, it would default to manual leveling. This is not so and gives the error messages and crash/reset I reported yesterday. If I explicitly define the probe as being manual (M558 P0), then it does do a manual mesh of the bed - both through RRF and the TFT. In other words everything works properly on the TFT if the M558 and M557 commands are set up correctly - even if a z-probe is not installed.

Your idea (and build) to only allow access to the bed leveling functions if the M557 command is present, only half works. Taking out this icon would prevent access to the other functions on this screen like checking out the probe functionality during installation and prevent access to bed leveling (with multiple z-axis motors).

I think the safest option would be to always enable access to the Bed Level functions when RRF is used - everything works well with and without M557 and M558 defined. You just can't have M557 without setting M558 first. The only enhancement I would suggest for the TFT would be to consider displaying the error messages on the TFT that I see with RRF when things are not set up correctly - but this is not critical. It would be good if we could get this stuff into the user manual or similar document.

There's probably a case to always enable the Bed Level functions with Marlin, Klipper, etc.

In the light of this, do you still want me to test the build you just sent me? I'm very comfortable with your last version.

Wombat37, 2022-06-16:
OK - it's been great working with and sorry it took so long :)

The big problem I have is finding out how a lot of this stuff is supposed to work - documentation is rather sparse.

radek8, 2022-06-16:
If you want, you can create documentation for RRF yourself.
There are not many active users with this system.
Previously, there were two who were interested in actively working on development, but they did not respond for about 3/4 of the year.

Wombat37, 2022-06-16:
If I write that - how would it be published? In the user manual?

If so, I have a ton of other tricks and notes on setting up RRF to work with the TFT (learned the hard way).

radek8, 2022-06-16:
Maybe there should be a link to the PDF file with instructions for FFR users on the title page, in addition to the installation instructions.
@oldman4U , what do you think? If Wombat37 prepares a PDF file with instructions for RRF users, will you place it on the front page?

Wombat37, 2022-06-16:
If you're serious, I could act as 'editor' of such a document and, hopefully, others could send me further content/ corrections that I could incorporate.

Should it be here or on TeamGloomy Wiki as @jaysuk suggests? 

Personally, I would like to do it here as it would reach more non-RRF users and may convince them to give it a try.

radek8, 2022-06-16:
https://teamgloomy.github.io/tft.html

Wombat37, 2022-06-16:
Looking at the teamgloomy page now, looks like there's a gap for the BTT TFT screens and this would fit in well there.

oldman4U, 2022-06-16:
> I could do that - particularly with help from someone like @jaysuk
>
> This is diverging from the topic of this issue so I should stop posting
> this stuff - I'll reach out to you on Discord @jaysuk

If needed - I can help

Wombat37, 2022-06-16:
Thanks @oldman4U - I appreciate someone with experience to look over this stuff so that I don't say anything too stupid. 

I don't mind putting in the effort.

closed 2022-07-06 by bigtreetech

jaysuk, 2022-10-11:
Just drop me a pm, that's the best way