bigtreetech/BIGTREETECH-TouchScreenFirmware #2539: [BUG] BIQU_TFT35_V1.2.27.x Unable to use TFT SD card，no sdcard detected

langyaWang22, 2022-06-12:
BIQU_TFT35_V1.2.27.x Unable to use TFT SD card，no sdcard detected.
BIQU_TFT35_V1.2.26  Can be used normally.

radek8, 2022-06-12:
You probably used an old FW, or an FW that's not for your display.
Use the master branch.

[image]

langyaWang22, 2022-06-13:
> 您可能使用了旧固件或不适合您的显示器的固件。 使用主分支。
> 
> [image]

> You probably used an old FW, or an FW that's not for your display. Use the master branch.
> 
> [image]

Tried "master" or failed to use SD card

langyaWang22, 2022-06-13:
> 

> You probably used an old FW, or an FW that's not for your display. Use the master branch.
> 
> [image]

Tried "master" or failed to use SD card

radek8, 2022-06-13:
> > @langyaWang22 , @adambrejniak
> > 
> > * What HW variant of the display do you have?
> > * Which WF file did you use?
> > * The problem is that the SD card is not detected or is the card detected but not read?
> > * If the previous FW card worked, do you remember from which date the last functional FW was?
> > * Did you also use the config.ini file and the folder with icons and fonts when updating? Did the update take place without error messages?
> 
> Now I have TFT35_V1.2.26 Apr 11 2020 - It is working. If I use latest TFT35_V1.2.27 - SD not working. But, firmware was uploading. Mayby this version use to much RAM?

So your display version is TFT35 V1.2?
Does that mean 3.5 "without EXP1 and EXP2 connectors?

adambrejniak, 2022-06-13:
> So your display version is TFT35 V1.2? Does that mean 3.5 "without EXP1 and EXP2 connectors? Currently, TFT35 has about 11 HW versions and you need to use the correct FW.

Yes, 3.5" without EXP1 and EXP2. I use BIQU_TFT35_V1.2.27.x.bin. This should be correct FW

langyaWang22, 2022-06-13:
[](url)> > > @langyaWang22,@adambrejniak
> > > 
> > > * 你有什么显示器的硬件变体？
> > > * 你用的是哪个WF文件？
> > > * 问题是未检测到SD卡还是检测到卡但未读取？
> > > * 如果之前的 FW 卡有效，您还记得上一个功能性 FW 是从哪一天开始的吗？
> > > * 更新时是否还使用了 config.ini 文件和带有图标和字体的文件夹？是否在没有错误消息的情况下进行了更新？
> > 
> > 
> > 现在我有 TFT35_V1.2.26 Apr 11 2020 - 它正在工作。如果我使用最新的 TFT35_V1.2.27 - SD 不工作。但是，固件正在上传。Mayby 这个版本使用了很多内存？
> 
> 所以你的显示版本是TFT35 V1.2？ 这是否意味着 3.5" 没有 EXP1 和 EXP2 连接器？ 目前，TFT35 大约有 11 个 HW 版本，您需要使用正确的 FW。

> @langyaWang22 , @adambrejniak
> 
> * What HW variant of the display do you have?
> * Which WF file did you use?
> * The problem is that the SD card is not detected or is the card detected but not read?
> * If the previous FW card worked, do you remember from which date the last functional FW was?
> * Did you also use the config.ini file and the folder with icons and fonts when updating? Did the update take place without error [messages?]([url]([url]([url]([url]([url](url))))))

> @langyaWang22 , @adambrejniak
>
> * What HW variant of the display do you have?
> * Which WF file did you use?
> * The problem is that the SD card is not detected or is the card detected but not read?
> * If [...]

> @langyaWang22 , @adambrejniak
>
> * What HW variant of the display do you have?
> * Which WF file did you use?
> * The problem is that the SD card is not detected or is the card detected but not read?
> * If [...]

closed 2022-06-13 by langyaWang22

langyaWang22, 2022-06-13:
> @langyaWang22 , @adambrejniak
>
> * What HW variant of the display do you have?
> * Which WF file did you use?
> * The problem is that the SD card is not detected or is the card detected but not read?
> * If [...]
[image]
[image]
[image]

It can be upgraded and downgraded normally. There is no error when writing firmware.

radek8, 2022-06-13:
here is the vere of August 2, 2021
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/5fd8a7bea797d893ae339a499c8c90b1ffa72486/Copy%20to%20SD%20Card%20root%20directory%20to%20update

langyaWang22, 2022-06-13:
> Can you try to upload older versions of FW V1.2.27 to find out when it got damaged? For example, here is the version from December 4. Can you test? https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/484ffc2d1cdbe4ba6fa970ee9b6d2953c27a6194/Copy%20to%20SD%20Card%20root%20directory%20to%20update

TFT SD card can be used

radek8, 2022-06-13:
It is well
Can you try another FW between December 4 and the current FW to find out when it broke?

langyaWang22, 2022-06-13:
> February 14? https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/0df78c5c52ac570e8c8eda2d78e2731f8a2e3873

no sd card detected.

radek8, 2022-06-13:
January 18?
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/5cb1201e6ed159c8bbb747f5741136e7a9f7f58e/Copy%20to%20SD%20Card%20root%20directory%20to%20update

langyaWang22, 2022-06-13:
> January 18? https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/5cb1201e6ed159c8bbb747f5741136e7a9f7f58e/Copy%20to%20SD%20Card%20root%20directory%20to%20update

TFT SD card can be used

radek8, 2022-06-13:
Maybe I found it.
Try on January 25 and then January 26

January 25
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/41afaf40249e1fb31d14e227678145ed594de91f/Copy%20to%20SD%20Card%20root%20directory%20to%20update

January 26
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/186eb548e617dba5f5ff9817cd3e0cb15e8bc017

kisslorand, 2022-06-13:
@langyaWang22 
Please check PR #2540

adambrejniak, 2022-06-14:
> > February 14? https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/0df78c5c52ac570e8c8eda2d78e2731f8a2e3873
> 
> no sd card detected.

Thank You, it's working :)

langyaWang22, 2022-06-14:
> @langyaWang22 @adambrejniak Please check PR #2540
> 
> Thx @radek for narrowing down the search.

Neither the firmware using "2315" nor the firmware recompiled by changing the "master" code can use the SD card.
no sd card detected.

langyaWang22, 2022-06-14:
> @langyaWang22 @adambrejniak Please check PR #2540
> 
> Thx @radek for narrowing down the search.
"2315" recompiled firmware can use SD card.

PR #2540 (2022-07-06): Fix  print from SD on TFTs without SD detect pin
On some TFTs there is no SD card detect pin. For those TFTs PR #2351 introduced a bug so SD card cannot be selected as a source for gcode files. This PR fixes it.

Fixes #2539