bigtreetech/BIGTREETECH-TouchScreenFirmware #2542: TFT50.  Cannot get firmware to update.  Cannot get Portrait Mode.  Please Help 🥺

mdaroff, 2022-06-16:
Hoping someone can help with this, I've tried all I can think of.

Attempting to update firmware on TFT50 (ultimately to enable Portrait Mode).  Cannot get firmware to update even to a pre-compiled .bin.  Not sure what I'm doing wrong.

I have attached the files that I have currently saved in root directory on SD card.  (.bin file, config.ini, TFT50 folder (from the "copy-to-SD" folder of the repository), reset.txt (per Portrait mode instructions to trigger screen calibration) )

Procedure:

- Insert SD card into TFT50 SD-card slot
- power on SK3+TFT50 via 24v PSU

When i boot up the TFT50 (connected to SKR 3), fonts update, icons update, config updates...but for some reason the firmware does not, and remains ".bin" rather than ".CUR"

Any ideas?  Thanks!

[TFT50.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8914416/TFT50.zip)

k4blammo, 2022-06-16:
additional detail.  8 gb SD card used.  Sd card was low level formated FAT32 (not quick format).  Card is otherwise blank other than the files included in the zip file in first post.

radek8, 2022-06-16:
What is the model of your TFT?
Don't happen to have a GD variant?

radek8, 2022-06-16:
Do not put the reset.txt file on the SD card along with the bin file. 
Use the reset.txt file only in the second step.

k4blammo, 2022-06-16:
I do not believe it is a GD variant.  See attached.

Also, same result without the reset.txt.  fonts, logo, icons, config updates...but firmware does not.  😔

[image]

radek8, 2022-06-16:
Yes it's a GD version!

[image]

radek8, 2022-06-16:
Bigtreetech signage is pathetic

k4blammo, 2022-06-16:
Well I'll be damned!  Will recompile and give it another go.  Fingers crossed 🤞

k4blammo, 2022-06-17:
@radek8 ... I owe you a beer. 🍻

closed 2022-06-17 by k4blammo