bigtreetech/BIGTREETECH-TouchScreenFirmware #2545: Chamber temperature sensing causes bed temperature display malfunction

dewhisna, 2022-06-18:
### Description

I am using the Octopus v1.1 board with the TFT35v3 display.  I have a thermistor connected and configured to the `TEMP_2_PIN` to monitor the chamber temperature, but no chamber heater (`HEATER_CHAMBER_PIN` is undefined).  This is just to monitor/display the chamber temperature even though it can't be controlled.

The chamber (and bed) temperatures display and function correctly in OctoPrint on a connected RPi, but the TFT35v3 display fails to properly alternate the bed and chamber temperatures.  It will sometimes stop displaying bed temperature at all (either actual or target), sometimes will show the chamber icon instead of the bed icon at times when it should be showing the bed icon, and will sometimes just show `0/0` for the temperature where it should show either the chamber temperature or the bed temperature, and generally misbehaves.

I am suspecting it's related to the fact that I have Marlin configured to read/report the Chamber temperature but not control it, since there is no chamber heater.

### Steps to reproduce

1. Configure Marlin for Chamber temperature sensor (am using a simple 10K thermistor since it's low temp compared to hotend):
```
#define TEMP_SENSOR_CHAMBER 4
```
2. Configure Marlin temperature sensor pin (leave `HEATER_CHAMBER_PIN` undefined):
```
#if TEMP_SENSOR_CHAMBER
  #define TEMP_CHAMBER_PIN		TEMP_2_PIN
  //#define HEATER_CHAMBER_PIN      P2_04   // Required heater on/off pin (example: SKR 1.4 Turbo HE1 plug)
  //#define HEATER_CHAMBER_INVERTING false
  //#define FAN1_PIN                   -1   // Remove the fan signal on pin P2_04 (example: SKR 1.4 Turbo HE1 plug)
```
3. I have tried with `HEATER_CHAMBER` enabled and disabled in the TFT35 config.  It fails either way with erratic behavior, but does it more often and quicker when enabled.  Try it both ways:
```
#define HEATED_CHAMBER 1  // Default: 0
```

**Expected behavior**
I expect it to correctly display the chamber temperature as my OctoPrint does.  And expect it to correctly show the bed temperature, both actual and target.  I would even be OK with it just not showing the chamber temperature at all, since I can see it via OctoPrint (which is why I've tried configuring the TFT35 both ways), but I need it to show the bed temperature correctly.  However, it fails either way.

**Actual behavior**
It gets into a mode where it stops showing the bed heater temperature (either current or target) and displays the chamber heater icon when it should be showing the bed heater icon.  When it gets like this, you can't see the current bed temperature anymore from the display (either on the information screen or at the temperature bar at the top on other menus).  Sometimes switching from touchscreen mode to Marlin mode and then back will reset it and let it show the bed temperature again, but it doesn't always work.  A hard reset will make it work again until it gets back out of sync and into this strange mode.

### Hardware Variant
TFT35v3

### TFT Firmware Version & Main Board Firmware details
TFT Firmware is at 1e5955e76d764e830ca707508052275d31f2e95e

Marlin Firmware is at https://github.com/MarlinFirmware/Marlin/commit/8aca38351cd6d33a1202bf9a13890dc8e5eadd49

### Additional Information
* Configuration.zip is attached with both TFT35 and Marlin config [Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8933939/Configuration.zip)

* At present, I don't have a picture available, since I haven't had a camera available at the time it occurred.  I will try to find a camera and get a picture the next time it happens.

dewhisna, 2022-06-18:
When it happens, it sometimes looks like this, where it says "Ch." instead of "Bed", yet is showing the icon for the Bed.  Sometimes, it's reversed and will show the chamber icon and say "Bed".  But the temperature it shows is `0/0`, which is wrong for both the Bed and the Chamber.  In this particular example, both the bed and the chamber are at room temperature at 18-19 degC.  And it also never cycles correctly through bed and chamber temperatures.  It stays showing the same wrong icon for the label and wrong temperatures.

[image]

In this particular example, the temperature at the top bar of other menus happens to appear correctly, as shown here.  But sometimes even it shows wrong values too:

[image]

If I switch to Marlin, you'll notice that it shows the temperatures correctly and always does:

[image]

And if I switch from Marlin mode back to touchscreen, sometimes it will fix itself, as it did this particular time (but not always):

[image]

radek8, 2022-06-19:
Do you have a printer with two nozzles?. 
Why does the second nozzle not show the temperature?

dewhisna, 2022-06-19:
I do, but it's showing no temperature because the second extruder is unplugged:
[image]
[image]
[image]

digant73, 2022-08-15:
@dewhisna All of your issues (icon and text misalignment, wrong temp 0/0 etc...) are due to the response sent by Marlin to command M115 (sent at startup by the TFT). In particular, I suspect Marlin is providing the capability value `Cap:CHAMBER_TEMPERATURE:0` (meaning no chamber is enabled/supported in Marlin). That value will force to 0 the param `heated_chamber` that you set to 1 in `config.ini` and triggers your issues. If you set to 0 the param `heated_chamber` in `config.ini` you should solve your issue (although you reported the issues are still present). Verify if Marlin periodically sends temp ACKs for the chamber (that ACK will also enable (set to 1) the param `heated_chamber` (but in this case you should not have any issue)).

closed 2022-08-19 by bigtreetech

Giraffa12, 2024-02-17:
Hi, I'm having the same problem. I have a TFT70 V3.0.27.x screen and a BTT skr 3 EZ mainboard with marlin V2.1.x. 
Is there any way to fix this error?

dewhisna, 2024-02-18:
@Giraffa12  -- I don't know.  I kept having problems with the Marlin firmware causing my printer to reset during long prints.  See MarlinFirmware/Marlin#24158.  I couldn't reliably get any prints that took a couple of hours to print.  Short prints were generally fine, but longer prints and it would just randomly reset and lose the print.

So, I ended up switching to Klipper and it's been running flawlessly.  It was a very different way of doing things, compared to Marlin, but I love it.  In fact, I'm in the process of doing some more rebuilding on the printer.  I'm switching from the Octopus down to a SKR 3 mainboard and will be putting EBB42 boards on the extruders with a CAN bus and get rid of 90% of the wires running to the extruders.  With it, you only need 4 wires -- two for power and two for CAN and even run multiple extruders with that.  It's much better than the mass of wires normally running to the extruders that flex during printing and break and cause issues.

Unfortunately, though, this display is quite limited for that configuration.  I have to run the display in "Marlin" mode as a dumb display device, since you can't send the "smart commands" to it via serial like you can with firmware like Marlin.  Instead, you need a display that can interface with the RPi that's driving the system.  (on Klipper, all of the control logic, calculations, movement commands, etc., is in the RPi with the mainboard of the printer acting as a dumb I/O port device).  Eventually, I'll probably switch this display out for a Pi TFT display, once I figure out a way to mount it to the printer (as it will be a slightly bigger display).

So, I don't know if this display bug still exists or not or how to solve.  But good luck with it!

PR #2577 (2022-08-19): Minor bugfixes + cleanup on configuration files
Some minor bugfixes + cleanup on configuration files.

fixes #2544
fixes #2545

**PR STATE:** Ready for merge.