bigtreetech/BIGTREETECH-TouchScreenFirmware #2546: "Click for summary" after printing -> screen freezes

AndylgTom, 2022-06-21:
TFT Freeze
-->

**Description**
At the end of printing, if you click "Click for summary" to view, the screen freezes. Only reboot helps. What could be the problem?

**Hardware**
Bigtreetech TFT 43 - SKR1.4 Turbo Board
USB flash

**TFT Firmware Version & Main Board Firmware details**
Bigtreetech Version last version
Marlin 2.0.9.3

**Additional Information**
Include a ZIP file containing marlin confug [Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8948888/Marlin.zip)

kisslorand, 2022-06-22:
Can you upload a gcode file that you used and had this issue with? If yes, please do so.

AndylgTom, 2022-06-22:
With any gcode this happens.... example (slice in prusaslicer)
[001-006.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8956250/001-006.zip)

kisslorand, 2022-06-22:
What happens if you go back where you select the media and press the "Brief" button?

AndylgTom, 2022-06-22:
Returned back, pressed "Brief" and the screen also froze.

kisslorand, 2022-06-23:
Can you test with the attached file? It is not extruding anything, safe to use on any printer setup.
[Lorem.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/8964902/Lorem.zip)

AndylgTom, 2022-06-24:
I checked. With this g-code, the "Click for summary" and "Brief" buttons on the screen work fine, the screen does not freeze.

rondlh, 2022-06-24:
I tried both files (stripped out the Gcode in the middle, left some at the start and end), but found no problems for "Click for summary" and "Brief", I'm using a BTT TFT35 V3.0

kisslorand, 2022-06-24:
My test file was to check if the time calculation is the culprit or not. It's not.

AndylgTom, 2022-06-25:
It also turned out that if you stop printing (forcibly) by pressing "Stop" and then "Click for summary" (or "Brief"), the screen even freezes too, even with your gcode. 

Maybe some option in marline is disabled? I don't know what else it could be :(

Video:
1. Ending print model https://disk.yandex.ru/i/BWDiQUXC6XHHvA
2. Stop https://disk.yandex.ru/i/DQKinYMT_6ZNkw

kisslorand, 2022-06-25:
It's hard to come with any suggestion as I cannot reproduce it.
When you updated your TFT's FW, did you also update all the other files too (config.ini, language_xx.ini)?

radek8, 2022-06-25:
@kisslorand 
The problem is in the Russian translation.
If you want to simulate a problem, get Russian.

radek8, 2022-06-25:
I used to have a problem with strange behavior in calibrating E steps.
The problem was also in translation. was caused by the national letter "ů"
After removing it, it was fine.

radek8, 2022-06-25:
Since Russian uses Cyrillic, it will be difficult to find which character and which expression makes this problem.

AndylgTom, 2022-06-26:
Yes, confirm, problems in Russian uses Cyrillic. Thanks [radek8](https://github.com/radek8). Hope you fix this.

AndylgTom, 2022-06-26:
Phew. I found the problem.
The problem is in the lines:
label_process_aborted:Process aborted!
label_process_completed!
Probably it happens not only here, but everywhere where label_process_...

If these lines are translated, then the screen freezes (i.e. a window with the corresponding inscription should pop up). It is also checked in Ukrainian.

radek8, 2022-06-26:
Replace the rows with this and it will work.
label_process_completed:Завершён!
label_process_aborted:Прерван!

AndylgTom, 2022-06-28:
Ок.
I tested... it turns out that the problem is not some letter, but too long translation. If the text is transferred to the second line, there is this problem. Will you fix this problem or not? Or will I have to shorten the translation?

radek8, 2022-06-28:
It is not a question of whether someone can correct the Russian translation, but whether someone will want to do it.
The civilized world supports Ukraine. Russia no longer belongs to the civilized world.

AndylgTom, 2022-06-29:
Why is politics here? Is it allowed by the rules?

I can update myself. I ask whether to fix the firmware so that I do not have to cut the translation.
And by the way, with the Ukrainian translation, the screen also freezes. I think not only in these 2 languages.

digant73, 2022-06-29:
@AndylgTom the issue could be related to a too short buffer for some languages. You can try to increase the buffer size from

`char tempstr[30];`

to

`char` tempstr[60];`

in **printSummaryPopup()** function on **PrintingMenu.c** file. At least, in case you abort a print, the freeze should be no more present.

kisslorand, 2022-06-29:
Mr. @digant73 is right. Cyrillic characters are not limited to 1 byte, so for example "Процес перерваний!" ("Process aborted!" in Ukrainian) has 18 characters but it uses 34 bytes. I would rather double the length of both variables.
```
char showInfo[300];
char tempstr[60];
```

PR #2567 (2022-08-02): Minor cleanup
Minor cleanup.

fixes #2546

**PR STATE:** Ready for merge
closed 2022-08-02 by bigtreetech