bigtreetech/BIGTREETECH-TouchScreenFirmware #2547: No Progress Reporting with RRF

Wombat37, 2022-06-29:
Somewhere along the way, the % Progress stopped being reported on my TFT35 E3 V3.0 display coupled to a BTT E3 RRF V1.1 main board running RRF V3.3.11. The displayed progress remains at 0% during the course of the print as seen in the screen capture below.

[image]

I'm trying to track down which version introduced this bug. As I have to make a print each time I do this test, it's taking me some time to get this done. So far I've tested about 8 revisions and found that the problem was introduced between Sept 23 2021 (works) and April 1 2022 (doesn't work). I will narrow down the version as I test further. If anyone wants to suggest a version to try, let me know.

Wombat37, 2022-07-18:
OK, I've tracked down the apparent version that stopped the progress being reported under RRF.

Progress is reported using version 11/18/2021 (#2250)

Progress stops being reported in version 12/1/2021 (#2260)

I can't download the intervening versions as they give me a 404 error when trying to access them. 

It looks like a lot of redundant settings (for RRF) were deleted from the config_rrf.ini file in the 12/1/2021 version. One of these settings could have been used as a flag to enable reporting progress.

I could start to selectively start restoring the configuration settings line by line but this would be a long process as I would need to set up a new print each time. It might be better to approach this from the code angle.

Hopefully, a quick workaround for this problem would be to restore one of the configuration settings. 

I would welcome guidance on what to check next.

digant73, 2022-08-14:
@Wombat37 The binary for PR #2260 (with other merged PRs) was merged on 11/29/2021.
Try the binaries in the following zip file and let me know which is working or not.

[BIGTREE_TFT35_V3.0_E3.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/9333341/BIGTREE_TFT35_V3.0_E3.27.x.zip)

Wombat37, 2022-08-16:
Hi @digant73,

Thanks for looking into this.

I've been away for a few days but will now start to test the files you sent me. I've started with the 20111118 version but it's asking me to update the fonts, icons and config files too. Which ones should I be using? Should I be using these for all the other tests? 

Thanks

Wombat37, 2022-08-16:
OK, in the interests of time, I tried to download these files. Looks like the code files are a bit picky about fonts, icons and config files.

I managed to get 20111118 running with fonts and icons from the Dec-01 pull request and the config_rrf.ini from the Nov-18 pull request :) I hope this is good for the tests you want.

digant73, 2022-08-16:
Retrieve the content (binary, config.ini, icons etc...) browsing the repo and retrieving only the versions reporting the description `update prebuilt binaries and config` (so the last update committed by bigtreetech).
Below the example to retrieve all the PRs released on 11-18-2021.

[image]

Wombat37, 2022-08-16:
I'm sorry but you must forgive me - I not very familiar with using this repository. To find previous builds, I've been looking at the closed Pull Requests. This is obviously not what I should be doing.

So where do I find the "repo history"? It's not obvious from the main GitHub page.

kisslorand, 2022-08-16:
@Wombat37 
Stay put, I am about to send you a TFT FW for testing based on the actual master .

Wombat37, 2022-08-16:
I assume you meant by "actual" master, the "current" master. I installed the latest build (15 days ago) and overwrote the binary with the file you supplied in your lost comment. I then ran the wait.gcode file from the RRF DWC webpage..

This gave an incrementing progress display from 0 to 100% and corresponding slide bar movement and ended with a "process aborted" message. I assume that this is what you were looking for.

Let me know what else I can do.

Thanks

kisslorand, 2022-08-16:
> I assume you meant by "actual" master, the "current" master. 

Yes.

> This gave an incrementing progress display from 0 to 100% and corresponding slide bar movement and ended with a "process aborted" message. I assume that this is what you were looking for.

Isn't this also what you were looking for?

Wombat37, 2022-08-16:
Sorry - yes I am but I don't know what's behind your test.

From your comment it looks like a fix is close.

Thanks

kisslorand, 2022-08-16:
Upon further testing It seems it is the RRF who gives the "Abort" message to the TFT.
The test gcode file was made just for testing the functionality of the progress.

As I said I am unfamiliar with RRF, so maybe something's missing from that gcode file that RRF expects. Maybe you can test it with a real gcode file... A really simple file would do it (ex one single line or circle).

Wombat37, 2022-08-16:
Ok - running a real test now - this is a 45 minute print but I should see the progress increment - I'll see if I get the abort message at the end.

I'll keep you posted.

Thanks for all your efforts - they're much appreciated!

kisslorand, 2022-08-16:
I am also thanking you for testing things. It's passed midnight here, so... tomorrow is another day.

Wombat37, 2022-08-16:
OK - have a good night!

In the meantime, all seems to be working properly - the progress is being reported correctly. I'll wait to see if the abort message appears and post back.

Thanks again

Wombat37, 2022-08-16:
OK - good news! 

Everything seemed to work properly - the % progress and the sliding bar all incremented during the print and no abort message appeared at the end.

Thank you so much for fixing this useful feature.

kisslorand, 2022-08-17:
Fix is up: PR #2580

Wombat37, 2022-08-17:
Looks like you had to make some significant changes to the code to get this working. I'm surprised it took this much work to get something working again that worked before :)

Thank you again for all these efforts. They are much appreciated.

PR #2580 (2022-08-19): RRF progress fix
### Requirements

 - BTT or MKS TFT

### Description

 - parse RRF print percentage message

### Benefits

 - those using RRF

### Related Issues

Fixes #2547
closed 2022-08-19 by bigtreetech