bigtreetech/BIGTREETECH-TouchScreenFirmware #2548: [Q] (Your Question?TFT35 V2 Flashed touch screen mode, but nothing but black screen with Marlin Mode ?)

Wick3dJack, 2022-06-30:
I flashed my TFT35 V2 with the touch screen option am I supposed to use Blocked Marlin mode or something because all I have is a black screen with Marlin Mode in white at the top center screen.
This has never happened before and how would I remedy this ?
Thank You

rondlh, 2022-06-30:
Did Marlin mode work before you flashed the firmware? Did you add the required wiring? (2x 10pin flat cables)

oldman4U, 2022-07-01:
What is the status of this ticket?

Please help the community and close the ticket once you do not need it anymore. Thank you

radek8, 2022-07-02:
@rondlh ,
TFT35 V2 does not have EXP1,2 connectors and Marlin mode.

Wick3dJack, 2022-07-04:
No the tft35 v2 doesnt even support marlin mode which is why I'm asking how did it get installed ? I never enabled it in the config file

radek8, 2022-07-04:
Did you use the correct FW?
Didn't you use the bin file for a different model?

radek8, 2022-07-04:
Save the empty reset.txt file on the SD card, insert it into the TFT and reset the display.

Wick3dJack, 2022-07-04:
No I used correct bin I always double check & tried the reset.txt already but I think that just resets the Marlin config.ini settings

Wick3dJack, 2022-07-05:
Well I just tried using FlyMcu to flash the BIQU_TFT35_V2.0_bootloader.hex from 2019-11-07 and now getting illegal user app on the screen =\

closed 2022-09-24 by stale[bot]