bigtreetech/BIGTREETECH-TouchScreenFirmware #2550: [BUG] Hangs in Loading when opening USB drive

marcosg-peya, 2022-07-03:
### Description

The screen hangs at loading screen when trying to see files of usb.

### Steps to reproduce

1. Connect pendrive
2. Select Print
3. Select USB Disk

**Expected behavior**
To see files of USB disk
**Actual behavior**
Infinite "Loading..." screen.

### Hardware Variant
BTT TFT35 B1 V3.0
### TFT Firmware Version & Main Board Firmware details
TFT: V27.x May 30 2022
MainBoard: Marlin 2.0.6

The USB drive its a Sandisk Connect of 32GB(trying to use it to upload the files over wifi).
Another USB Drive works as expected(64Gb), maybe something special with the Sandisk one?.

There is no output on the serial port 2 (I have ESP3D there)
Is there any way to know what is the board doing that hangs or takes so long?

radek8, 2022-07-04:
What format is your disc?
It should be FAT32.

kisslorand, 2022-07-04:
The Sandisk Connect needs more power than the TFT can provide. It's the same story if you would like to use a WiFi SD Card like BTT TFT Cloud.

marcosngomezi, 2022-07-04:
Hi guys, its formatted in fat32, current problem makes sense, how much is the limit of the tft? Is there any alternative?

Sandisk Connect needs to be ejected to be able to upload via wifi, I understand that the tft also don't supports ejection right?

I was thinking to do an extension board that could eject the usb and now it should give extra current also.

There are some IO pins available for custom things right? which ones? are they handled with custom G-codes?

closed 2022-07-08 by marcosngomezi

marcosngomezi, 2022-07-30:
Hi, @kisslorand and @radek8, FYI after some debugging I found out that the problem was that de scsi command modeSense6 was failing, fallbacking to write_protected on failures of the modesense the TFT was able to read it without problems