bigtreetech/BIGTREETECH-TouchScreenFirmware #2557: Erroneous control of extrusion distance on TFT movement screen with RepRap

Wombat37, 2022-07-14:
This was a bug reported and fixed as #1909 - it seems to have returned in the current version.

I believe that it's an issue with absolute versus relative extrusion from the TFT extrusion screen.

kisslorand, 2022-07-19:
Check if PR #2560 fixes the issue you have.

Wombat37, 2022-07-19:
Many thanks - I will test this later today.

Wombat37, 2022-07-19:
OK - I tested it. It still doesn't work - the extruder moves by the total displayed rather than the incremental adjustment. This applies when to both pressing buttons and rotating the knob.

kisslorand, 2022-07-19:
I made a little change, can you please try again?

Wombat37, 2022-07-19:
OK, tried again but it's still not working.

Am I using the right link to get the test build?

https://github.com/kisslorand/BIGTREETECH-TFT/tree/G90/91-Fix

Wombat37, 2022-07-19:
Did I need to recompile the code for this test?

kisslorand, 2022-07-20:
> Did I need to recompile the code for this test?

Yes

Wombat37, 2022-07-20:
OK - sorry. I'm a user not a development programmer and sometimes I need to be given directions.

Anyway, I compiled your latest code with the 'little change' and ran it on my RRF system.

The extrusion distance was still not working properly from the TFT display..

Wombat37, 2022-07-22:
@kisslorand, many thanks for your help with this - it's much appreciated. 

What else can I do to help get this fixed?

kisslorand, 2022-07-22:
I didn't gave up just had no time for it.
I do not have RRF printers nor access to one so I am poking in the dark.
Can you please run `M82` just before entering into "Extrusion" menu?
It would help to figure out is it an absolute/relative extrusion problem or a wrong "E axis" coordinate.

Wombat37, 2022-07-22:
Yes - the M82 entered beforehand made it work properly.

Wombat37, 2022-07-22:
I'll compile and try your new code

Wombat37, 2022-07-22:
Sorry - that latest version didn't work properly either.

kisslorand, 2022-07-23:
Sure. I attach 2 test files. Please test both

1. https://mega.nz/file/Lk5WlKhQ#LXD6pjjeeWg_c-y1MKHYUWHAHo8lHs4m1Z2JrzanSkM
2. https://mega.nz/file/SgxQTRIC#IIjnAOtgDM3vFofnMyOPETk4h1rJdJBrL0cWDa-oU-A

Wombat37, 2022-07-23:
Thank you for your continuing efforts to get this fixed :)

OK, I just tested both of these new versions on my E3-RRF V1.1/TFT35-E3-V3.0 system and neither works as it should.

They still extrude the filament according to the total on the display rather than the incremental movement expected. This gives two main problems:

1. Huge lengths of filament are extruded after just a minor turn of the control knob
2. Once forward movements are made, the knob has to be rotated right back to zero before negative movements can be made (and vice versa).

Looks like I was probably compiling the code correctly

So, we still haven't found a fix. Hopefully the M82 result is a big clue?

kisslorand, 2022-07-23:
Yes, the `M82` result and the results of those 2 binaries you tested are a big clue. It means to me that although the TFT upon startup sets the extruder to absolute mode, something (probably external) sets it back to relative.

Wombat37, 2022-07-23:
There was no external device connected during these tests (apart from the E3-RRF main board). DWC was not active.

kisslorand, 2022-07-23:
I do not have a single clue what sets your printer's extruder to relative mode without the TFT knowing it.
Can you find which commit broke it?

In the meantime I made another file for your TFT. It's not supposed to fix anything, it is just a test to check if the extruder is really set to absolute or not when the TFT boots up. With this FW (Test3) upon booting your printer is supposed to home X and Y and than Z. Please test it and report back.
https://mega.nz/file/ygo1gaaC#br2_vdNEU3mVLTcOxf0g4ODMVjOLvte4HWFeD5b-jjc

Wombat37, 2022-07-23:
Ok - will do. However, it’s 36C here at the moment and we have no AC - operating a 3D printer is quite a challenge😀 

I will check the new code you sent me very soon but it may take a day or two to find out which version broke it.

Thanks

Wombat37, 2022-07-23:
OK, loaded the new code (Test3). 

On boot, there is no homing occurring.

Wombat37, 2022-07-24:
PROBLEM MAY HAVE BEEN FOUND!

I'm trying to understand how this relative versus absolute manual extrusion should work.

Surely, we want relative extrusions to apply when using the load and unload functions. The RRF config.g file sets M83 in the second line of the config.g file to do this.

M82 will set the extruder to absolute mode - so why did the earlier experiment setting M82 make the TFT driven extrusions work (properly) in relative mode?  

I also checked to see how DWC controlled manual extrusion with the latest master build (July 12th) - and this worked well, so the RRF code is still working as it should.

Have we got the M82 and M83 functions somehow reversed?

For reasons I don't fully understand, I changed the M83 (relative extrusion) to M82 (absolute extrusion) in the RRF config.g file.

Now **everything worked** for manual extrusion - the TFT knob, the TFT buttons and DWC.

So, it would appear that DWC doesn't seem to care whether M82 or M83 is set in the config.g file and the TFT seems to care but has things backwards. I don't understand what's going on here but perhaps this makes more sense to you. Note that M83 has always been set in my system and, at one time, this worked with the TFT.

Hopefully, somewhere in all this, there's a good solution. I think we are homing in on it and it looks like there's no point in testing further code.

Is there ever a need for absolute extrusion control? 

Does this work properly with Marlin?

Wombat37, 2022-07-24:
Note also that RRF handles the G90 and G91 commands differently from Marlin (they don't affect the extruder).

kisslorand, 2022-07-24:
@Wombat37 
Thank you for your help, your tests made me get closer and closer. I had to install RRF to go further and... I found the bug.
Please undo the changes you've done in your "config.g" and test the TFT FW I attach (Test4).

Test 4: https://mega.nz/file/mwADySqJ#L6qHjny5Mqmapgqwa65KDGSRvxEKzAq92HaWcoU_7pk

> Note also that RRF handles the G90 and G91 commands differently from Marlin (they don't affect the extruder).

[This](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2560/commits/f3380f6f6497dec23524ce0e359659ce17c30b64) commit does exactly that.

Wombat37, 2022-07-24:
Yes!!!

**Test 4 appears to work very well** - the TFT rotary encoder and the load/unload buttons now control the manual extrusion properly. DWC continues to work properly, too.

Looks like you've cracked it!

Many thanks for your efforts.

Do I need to test any other functions or is the impact of the change you're making, localized?

P.S. If, in future, you need to check out any other RRF-related changes on a working system, I will be pleased to help if I can.

kisslorand, 2022-07-24:
> Do I need to test any other functions or is the impact of the change you're making, localized?

As I said, I am not familiar with RRF. This fix also makes the RRF motherboard to report to the TFT the settings from "config.g". I do not know what does it bring to the table, if it fixes any other issues or not. Maybe check the progress bar issue... (#2547)

Wombat37, 2022-07-24:
Yes - I'll check the progress bar issue. It would be wonderful if your changes fixes other stuff, too.

I'll report back in both threads if it does.

Wombat37, 2022-07-24:
This did not fix the progress display/bar issue (#2547).

closed 2022-07-29 by bigtreetech

Wombat37, 2022-10-11:
Does #1962 offer any clues?

kisslorand, 2022-10-11:
> Does #1962 offer any clues?

Is this addressed to me? If yes, clues about what should PR [#1962](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1962) offer?

Wombat37, 2022-10-31:
> > Does #1962 offer any clues?
> 
> Is this addressed to me? If yes, clues about what should PR [#1962](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1962) offer?

Sorry - I was travelling and not able to access a PC. Somehow I must have replied. Please ignore this and my previous message :)

PR #2566 (2022-07-29): RRF setupMachine Fix
### Requirements

BTT or MKS TFT

### Description

After PR #2171 the setupMachine() procedure was prematurely interrupted for RRF resulting in unknown absolute/relative mode of XYZE coordinates, no report of "config.g" settings and the LED not being initialized.

This PR fixes these.

### Benefits

RRF

### Related Issues

Fixes #2557