bigtreetech/BIGTREETECH-TouchScreenFirmware #2559: BTT SKR Mini E3 V3-  Ender6 example incorrect drivers

nicwilson58, 2022-07-18:
### Description

BTT SKR Mini E3 V3-  Ender6 example incorrect drivers,  they are listed as TMC2208_STANDALONE which cannot be correct

### Steps to reproduce

See Configuration, h for above example

**Expected behavior**
Drivers should be configured for listed board in the config not creality board. Expected TMC2209

**Actual behavior**
Wrong Drivers listed in configuration.h

### Hardware Variant
N/a
### TFT Firmware Version & Main Board Firmware details
N/a

### Additional Information
N/A

oldman4U, 2022-07-18:
Sure, you have been posting this using the correct repository?

oldman4U, 2022-07-21:
Please help the community and close the ticket once you do not need it anymore. Thank you

nicwilson58, 2022-07-21:
you have have not fixed the issue I checked today and it's still wrongSent from my Galaxy

oldman4U, 2022-07-21:
This is open source, so all users including YOU can change it. Furthermore, your issue does not belong to the Touchscreen firmware, but to the mainboard. So this is the wrong repository and you have to post the issue using the correct repository.

nicwilson58, 2022-07-21:
hi,ok I will fix it   in future if you do not want motherboard  config files in the TFT repository then do not put them there.  I shall fix. delete and move them to the correct place.Sent from my Galaxy

oldman4U, 2022-07-21:
Please do so. Thank you

nicwilson58, 2022-07-21:
completed.

closed 2022-07-21 by nicwilson58

nicwilson58, 2022-10-11:
the examples are in this repository just look its under ender6Sent from my Galaxy