bigtreetech/BIGTREETECH-TouchScreenFirmware #2561: SKR 3 EZ

Deanxr5, 2022-07-21:
Does the TFT 35 E3 V3.0 support the new SKR 3 EZ board?

I've been trying to get it working on klipper with no success. All that happens is a black screen. It does have a red LED on the back so I assume its got signal.

kisslorand, 2022-07-24:
> Does the TFT 35 E3 V3.0 support the new SKR 3 EZ board?

Yes.

Deanxr5, 2022-07-25:
Thanks, it's currently not functioning for me. I've used the pin layout found on github. Using EXP1-EXP2 even tried with the tft connection. I can only assume from forums I've read it need to be reset.

oldman4U, 2022-07-25:
You can find everything related to this repository and the BTT TFT firmware in the ReadMe which you can find at the main page. This is not a user group. 

Please help the community and close the ticket once you do not need it anymore. Thank you