bigtreetech/BIGTREETECH-TouchScreenFirmware #2562: [BUG] Icons: update required

mlaci001, 2022-07-21:
### Description

### Steps to reproduce

kisslorand, 2022-07-24:
You should read again more carefully the readme section from here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware. It is explained in great detail which file is needed and each of them goes where on the SD Card.

mlaci001, 2022-07-25:
> You should read again more carefully the readme section from here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware. It is explained in great detail which file is needed and each of them goes where on the SD Card.

Thanks.
I finally found the solution.
I formatted the SD card with a block size of 2048 instead of the default one, and so not only the firmware and fonts were updated, but also the icons.
The content on the card has not changed (folder structure, contents of folders, etc.)

oldman4U, 2022-07-25:
Perfect. Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2022-07-25 by mlaci001