bigtreetech/BIGTREETECH-TouchScreenFirmware #2564: [FR] Sound support for host commands (M300)

rondlh, 2022-07-24:
The TFT can play sounds when a gcode command (M300 TFT) is received from the secondary serial port, but not when the same command is received from the host. This is a pity because Marlin can easily be configured to forward sound commands to the TFT via the ExtUI interface. Adding M300 support in parseACK.c would be useful in my view.

Also, I noticed that the TFT uses interrupts to generate the tones, while using the hardware PWM would be the way to go. I understand that some poorly designed hardware could be the reason for this as some TFTs use the same PWM channel for the TFT back light and speaker. Not all TFT have this issue.
Or is there another reason to not use hardware PWM for the sound?

digant73, 2022-08-11:
what's the ACK message for M300 sent by Marlin?

oldman4U, 2022-08-22:
rondlh. Please answer digant73 question or close the Ticket. Thank you

oldman4U, 2022-09-02:
@Msq001 You may want to check and close this ticket. Thank you

guruathwal, 2022-09-13:
This question has been answered earlier here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1702#issuecomment-796842015
The ExtUI itself run on the same processor as marlin hence it is able to control the buzzer, While this firmware used Serial communication. Marlin does not forwards M300 commands through serial.

rondlh, 2022-09-18:
> what's the ACK message (related to M300) forwarded to TFT by Marlin?

Sorry for the late reply.

I let the ExtUI just forward the M300 command:
```
  void onPlayTone(const uint16_t frequency, const uint16_t duration) {
    char buffer[18];
    sprintf(buffer, "M300S%dP%d\n", frequency, duration);
	PORT_REDIRECT(SERIAL_PORT_TFT);
	SERIAL_ECHO(buffer);
  }
```

Then in the TFT parseACK.c I pickup the M300 and execute it locally.
I hope this answers the question.

closed 2022-11-04 by rondlh