bigtreetech/BIGTREETECH-TouchScreenFirmware #2565: [FR] Terminal enhancements, enable editing

rondlh, 2022-07-24:
Here some ideas to enhance the capabilities of the terminal:
1. Remember the last command, this makes repeating the same command easy.
2. Allow editing of the command (backspace support only), so you can easily go from M500 to M502 for example.
3. Erase the whole command by pressing the box that shows the command (top left), this is useful so you don't have to backspace the whole command.
4. Show the digits in a different color (yellow)., this makes the terminal keypad look much nicer and easier to use
[image]
.

kisslorand, 2022-07-24:
> 2. Allow editing of the command (backspace support only), so you can easily go from M500 to M502 for example.

"Del" button does that.

rondlh, 2022-07-24:
You are right, unfortunately it can only be used to correct a typo, not to change a previous command

PR #2575 (2022-08-19): Added short gcode history to Terminal menu
**IMPROVEMENTS:**
* Improved Terminal menu:
  *  Added an history of the last 3 sent gcodes. It allows to load a previous sent gcode from the history instead of re-typing the whole command from scratch.
  *  Added a `Clear` button. It allows to clear the current displayed gcode instead of clicking multiple times on `Del` button.

The gcode text box (top left area) in the Terminal keyboard menu provides 3 clickable buttons:
* `Previous` (left button): it scans the history from the last (recent) to the older sent gcodes.
* `Next` (center button): it sca [... cut]
closed 2022-08-19 by bigtreetech