bigtreetech/BIGTREETECH-TouchScreenFirmware #2569: [Q] BTT E3 RRF V1.1 & TFT35 E3 3.0.1 Issue or wrong config?

Zarrsito, 2022-08-05:
Hey i just installed the TFT35 display as mentioned in the Title, im running on a BTT E3 RRF V1.1 using latest RepRapFirmware
i can control my Printer just fine, but when i pause of the WebInterface the Display not really reckognize it, when i go to printer to resolve the problem why i paused and try to start from display it still says pause, when i hit that button gives me an error its already paused, and i can continue then...

And sometimes it does not apply the Z offset from my Probe just heading to z+5 without adding the offset ending up printing in air sometimes... any ideas? =) power off/on resolves this behaviour...

Customized Ender 3 

Added this as Question since maybe im only Dumb and cant see the Forest because of the Trees xD

Zarrsito, 2022-08-22:
nice seems nobody has an answer... well
i noticed that the z-offset or any other settings are stored to the eeprom... well it gives me no error, but when i turn off printer and turn back on, the display lost all settings, z-offset and what is set... did i miss something?!?!? and i tryed now different settings but nothing helps against the pause bug.... even when i pause from display it doesnt reckognize it... need to hit pause several times to be able to hit continue... im only connected over the serial connection since i dont use marlin mode or is that the problem!?!?!?

radek8, 2022-08-27:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2572#issuecomment-1217818116