bigtreetech/BIGTREETECH-TouchScreenFirmware #2571: SD card read error.

wigw, 2022-08-08:
TFT70 cannot see USB files, crash when printing from SD card - message: SD card read error.

rondlh, 2022-08-12:
What file format does the SD card use? What size of card it it?
The format should be FAT or exFAT, sometimes large cards give problems.

wigw, 2022-08-12:
Hi, I'm using the standard SDcart from the BTT box.
Today, at times I printed from SDcart, USB sees and opens, but for some reason the directory is empty.
I don’t know what to do, please help.

radek8, 2022-08-13:
Are you using the current FW on the TFT?

wigw, 2022-08-14:
Yes, I'm using the latest release of FW.

digant73, 2022-08-14:
@wigw Try to use also a different SD to micro SD adapter and to clean the contacts in the SD slot.

Wombat37, 2022-08-16:
I'm seeing similar problems with RRF. Sometimes the TFT (TFT35-E3-V3) sees the stored files on the TF card (E3-RRF-V1.1) and sometimes it shows an empty screen and sometimes it locks up. Other times it works properly.

I have no problem downloading and running a print file remotely from the RRF DWC, so the TF card is working properly.

It seems to change when I upload different TFT firmware versions.

Wombat37, 2022-08-19:
Hi @digant73 , thank you for following up.

I took a look at the TF card and realized that I had 167 files in the gcode folder occupying about 500 MBytes. I then archived most of these to a backup folder on my PC thus reducing the number of gcode files on the TFT card to just 10. These files were now displayed and could be selected from the TFT display.

It looks like the TFT firmware may have an issue with a large number of files or a large memory storage.

The work-around is of course, to limit, how many files get into the gcode folder. I can live with that.

If there is a limit to what the TFT can handle, perhaps think about posting a message rather than just showing a blank screen and locking up :) The RRF DWC can obviously handle this number of files.

Otherwise, I'm now good - thank again for reaching out.

digant73, 2022-08-19:
@Wombat Yes, the file browser in TFT fw allocates memory to hold the file names of all the files present in the target folder being scanned (e.g. root folder). For that reason I had the feeling that you provided too much files in the same folder (you mentioned you had some freezes during the browsing).
There is no limit on files number. The allocated buffer basically depends on files number and each file name length present on the scanned folder.

wombat, 2022-08-19:
@digant73 I think you meant @Wombat37 😊

Wombat37, 2022-08-19:
OK - I think we're good.

Sorry to raise another moot issue but it's good to know what the potential problem is and how to fix it.

Thanks again

wigw, 2022-08-19:
Hi grupa. I'm far from the printer now. I have one usb stick in the extfat format, the second usb in the fat format. Neither the first nor the second usb stick tft sees. Maybe I'll take a photo in a couple of days.

wigw, 2022-08-19:
Micro SD fresh gcode reads, I made some good prints.

wigw, 2022-08-20:
Hi, today I took a photo of a usb directory
[image]

wigw, 2022-08-20:
(https://drive.google.com/file/d/11h1g_usYNeWB_ARqkjSZqU4RU3wXmTxL/view?usp=drivesdk)

digant73, 2022-08-21:
@wigw no issue at all with those files on my USB stick (FAT format). You now reported SD card is working fine. What was the solution? New more reliable micro SD card, new SD card adapter?

wigw, 2022-08-21:
Hi, I used the same tf card and old adapter, just created a new gcod.
USB does not open, I re-formatted and burned gcod, still usb does not see.

digant73, 2022-08-22:
Make sure you properly unmount the USB stick from your laptop/PC

wigw, 2022-08-22:
Yes, I always correctly disconnect from the PC

wigw, 2022-08-23:
[image]
I switched

wigw, 2022-08-23:
[image]
I switched, the result is the same

digant73, 2022-08-23:
@wigw Try to remove the folder named `overlays` (provide only the files in the root folder)

wigw, 2022-08-23:
I don't see this folder in Windows Explorer, there are only two .gcod files in the root of the folder.
I don't know how to delete it.

digant73, 2022-08-23:
Are you using Windows or Linux to format the USB stick. It seems to me you are using a sort of protected fs or you have a virus in your laptop/PC

wigw, 2022-08-23:
Perhaps you are right, but this is not the case with tf-cart. I formatted, wrote down two .gcod files, you saw the result. Tomorrow I will be at work, I will check the usb with an antivirus.

digant73, 2022-08-23:
check all your laptop/PC with an antivirus. Try also to format the USB stick from another laptop/PC (and also copy the files from a different laptop/PC).

wigw, 2022-08-26:
Hi guys.
Thank you all I withdraw my question. The problem was my tf card and usb being tired. The full and correct format fixed the situation.
Many thanks to everyone for the support.

closed 2022-08-26 by wigw