bigtreetech/BIGTREETECH-TouchScreenFirmware #2572: [FR] Send Pause and Stop Command to While Remote Printing

Rneil-ca, 2022-08-09:
Would be really nice to be able to use the pause and stop key when remote printing.  This could be set up to send a M601 that would pause Octoprint with the right config.

Thanks,
Ryan

digant73, 2022-08-11:
AFAIK M601 is already handled by the TFT fw.

digant73, 2022-08-17:
With remote printing engaged, the TFT has no idea of the host managing the print (it can be OctoPrint or any other unknown client) so the only rule that can be followed by the TFT is simply to act as passthrough between remote host and mainboard. Furthermore, the TFT does not support response forwarding for request originated by the TFT (e.g. M601). Even if the TFT supported the response forwarding, the TFT couldn't make any assumption about the handling provided by the remote host receiving the response (e.g. the remote host could discard the notification and continue the print).

For those reasons, the TFT's role is limited to display stats info and nothing more.

closed 2022-11-07 by stale[bot]