bigtreetech/BIGTREETECH-TouchScreenFirmware #2574: [Q] Use USB to communicate with printer

xhejtman, 2022-08-12:
**Your question or details you need. Please describe.**

Is it possible to use USB cable for communication with printer instead of serial cable?

kisslorand, 2022-08-16:
If you mean to connect the USB of the TFT to the USB of the mainboard, than no, it will not work.

xhejtman, 2022-08-16:
OK. To explain: I have SKR 3 controll board and TFT display, I used Marlin, however, there are some serious issues with UARS. TMC2209 drivers cannot connect and print over serial from the TFT freezed many times (not in the case of print from internal SD card from controll board). So I was hoping that connecting via USB would solve freezes. But in the end, I moved to klipper and everything works now.

closed 2022-08-16 by xhejtman

kisslorand, 2022-08-16:
Good to hear you found a way! Happy printing!