bigtreetech/BIGTREETECH-TouchScreenFirmware #2583: How to fix vscode not compiling new firmware for touchscreen?

jburkhardt0, 2022-08-20:
Hello

I am having an issue where I am not able to build/compile the firmware for my tft35 v2.0. 

Background: Want to edit the existing firmware for the tft35 v2.0 to allow for vertical portrait mode. 

Process: following the instructions found on this site to activate portrait mode, I opened VSCode, opened new folder, selected the "BIGTREETECH-TouchScreenFirmware-master" folder I unzipped, edited the platformio.ini to allow for portrait mode, saved platformio.ini, hit "build," command box opens at top of screen which has "Platform.io: Build" shown, hit enter, nothing happens. 

Anyone have a thought on what may be happening? Thanks in advance

digant73, 2022-08-21:
To compile, press the keys `CTRL-ALT-B`

jburkhardt0, 2022-08-21:
> To compile, press the keys `CTRL-ALT-B` (on Windows) or `CTRL-Option-B` (on MacOS)

When hitting CTRL-ALT-B, it does the same thing as if I hit build, brings up the command line at the top of vscode, and I hit enter, command line goes away and nothing happens

digant73, 2022-08-22:
Did you install PlatformIO extension in VSCode? It is required for such kind of project

jburkhardt0, 2022-08-22:
> Did you install PlatformIO extension in VSCode? It is required for such kind of project

Yes, PlatformIO is installed. I have used VSCode to create and install firmware for four of my printers within the last few weeks so it was working then, just not sure why it doesn't work for the screen firmware

digant73, 2022-08-22:
Ok, if you are familiar with VSCode and it's not working now, try to reinstall VSCode and PlatformIO. Probably something went wrong in a previous VSCode update.

jburkhardt0, 2022-08-22:
> Ok, if you are familiar with VSCode and it's not working now, try to reinstall VSCode and PlatformIO. Probably something went wrong in a previous VSCode update.

uninstalling and reinstalling seemed to do the trick, Thanks!

closed 2022-08-22 by jburkhardt0