bigtreetech/BIGTREETECH-TouchScreenFirmware #2584: [Q] (Your Question?)

tttttim5, 2022-08-24:
**Your question or details you need. Please describe.**
Hello,

I wanted to add a feature so as soon as the TFT is connected to the mainboard, the temperature would be set to a predetermined value. This feature does already exist for Octopi, but before Octopi is started up, some time has been passed while the TFT is earlier connected. 
My question is in which file or in which piece of code does the TFT go through only once when it connects to the mainboard? Is this inside 'parseACK.c', 
`
if (infoHost.connected == false)  // not connected to printer
    {
...
infoHost.connected = true; // THIS LINE THAT IT IS CONNECTED?!?
requestCommandInfo.inJson = false;
}
` ?

Thanks in advance!

**specify the device variant related to your question.**
It concerns the BTT TFT, general software. I use the TFT35 E3 V3.0

**Additional context**
Nothing to expand on.

kisslorand, 2022-08-24:
Yes, that's the line in question.
You can insert your needed code wherever after the `goto parse_end;` line of the "if (infoHost.connected == false)" conditional in "parseACK()".

oldman4U, 2022-08-25:
Please update the title of the ticket and close it once your question is answered. Thank you

closed 2022-08-31 by tttttim5