bigtreetech/BIGTREETECH-TouchScreenFirmware #2588: [Q] BIGTREETEH TFT24 not flushed with custom firmware.

ZeriAD, 2022-08-26:
Question.

Hello.
I used a translation program because I am not good at English. It is difficult to deliver information, but I will tell you the current situation.

The BIGTREE_GD_TFT24_V1.1.27.x.bin Firmware file provided after purchasing BTTT TFT24.

After adding code from the provided open source to add the required features, I tried to compile and upload normally, but it did not flash.

Previously, in the case of BTT TFT35, it was flushed in all situations.

What more information should I provide to solve this problem?

Device Variant
Device : BIGTREETECH TFT24-V1.1
MCU : GD32F105 RCT6

radek8, 2022-08-26:
Use an SD card that is 8GB or less.
Format the card again to FAT32
If it doesn't help, try another card...

oldman4U, 2022-08-26:
Sure GD MCU is supported?

radek8, 2022-08-26:
Yes. Must use GD versions

oldman4U, 2022-09-02:
What is the status of the ticket? Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2022-09-05 by ZeriAD