bigtreetech/BIGTREETECH-TouchScreenFirmware #2589: Bin file not flashing on TFT35-E3

Wmelonhead, 2022-08-27:
Hello,

I bought an SKR Mini E3 V3 and a TFT35-E3 on Amazon for my Ender 3.  I have compiled the Marlin firmware correctly, just like my other Ender 3.  I have connected the Serial cable correctly and the other ribbon cable correctly.  I have followed every step on the README file for the TFT firmware.  I have spent many many hours trying to get the screen to work.  At one point, I had the touchscreen working normally, but the Marlin Mode would only display a blank screen with the words "Marlin Mode."  

The issue that I have had from the beginning and I now realize is the problem is that the .bin file will not flash.  I am using the BIGTREE_TFT35_V3.0_E3.27.x.bin file.  I have also tried compiling it myself following the CrossLink video.  I am using an 8gb sd card that is formatted to fat32 and 4096 bytes.  I transfer the TFT35 file, the config.ini file, and the .bin file to my sd card.  When I try to flash the firmware for the TFT, the screen does not load the .bin file, but it loads the other files.  When it is done, the TFT file and the config.ini files turn into .CUR files, but the .bin file stays the same.  I've even tried renaming the file firmware.bin.  I have also used three different 8gb sd cards, including one that is brand new.

I used to be able to use reset.txt to do a hard reset, but now even that will not work/flash.  My baudrate is also set to 250000 in both my Marlin firmware and the config.ini file and I have also been able to change it from the touchscreen mode.  But now, I can't get into the touchscreen mode.

Please help me.  I'm out of options.  I need the help of people much more knowledgeable than me.  I'm new to 3d printing.  Is my bootloader broken?  Is there a way to fix it?  Is my screen defective?  Or is there some other issue that I can resolve?

Thank you in advance.

kisslorand, 2022-08-27:
Try with "BIGTREE_GD_TFT35_V3.0_E3.27.x.bin".

Wmelonhead, 2022-08-27:
Thank you for the reply.  That did flash the bin file.  However, the screen is still not working.  If I click on the touchscreen mode, it flashes to to screen for a couple of seconds, where it says "printer not attached" and then it goes back to the screen where I choose the mode.  If I click on the Marlin Mode icon, it does the same.  It flashes to Marlin Mode, where everything looks normal, but only for a couple of seconds.  Eventually after I keep clicking on these icons, I hear a loud, constant beep and the screen will freeze where I won't be able to click on anything.  So I guess the bootloader is okay?  There is just something wrong with my bin file?  Maybe I'll try the other file right now and see if it works now?

Wmelonhead, 2022-08-27:
I just tried to flash the original bin file and it still doesn't load.

radek8, 2022-08-27:
According to the description of your problem, you have the display embedded in some kind of frame (for example, the original display holder on the Ender 3).
This frame touches the touch layer and presses on it.
Remove the display from the frame and try it.
If that solves the mode switching problem, use spacers.

radek8, 2022-08-27:
You must use the correct FW version as Kisslorand advises!
If the processor is marked with the letters GD on your display, you must use the bin file variant with this index.

Wmelonhead, 2022-08-28:
Wow!  You guys are amazing.  Thank you!  So simple.  I wish I'd posted this question a couple of weeks ago.  I understand what you are saying about the GD file and the processor.  That is obviously the file I needed and I've been trying to flash the wrong file this whole time.  And yes, I have my screen in the Ender 3 mount.  I had read about the screen hitting, but someone else suggested to loosen the bolts, so that's what I had done.  But it works now that I have removed the screen from the mount.  I will make some spacers to put it back into the mount.  

There is one other thing still not working properly and I bet you guys have the fix for that too.  In Marlin Mode, there is no back button.  I cannot go back to the menu or to the previous screen once I am in a screen like x movement.  How can I get that to work?  And thanks again!

radek8, 2022-08-28:
Yes, the missing back button is the current bug in Marlin.
It is already ready for repairs and waiting to be merged.

https://github.com/MarlinFirmware/Marlin/pull/24694

Wmelonhead, 2022-08-28:
I knew you guys would know the answer!  Thanks radek8!  And thanks kisslorand!  My back button works now.  Another easy fix.  And finally, my touchscreen works completely and I can finally start calibrating my printer.  Truth be told, I had bought another touchscreen for my other Ender 3 a while ago and I gave up on trying to get it to work and I just put the stock display on it.  Now I can go back and get that one to work.  

Thank you again for all of the help!  I was at the end of my road.  I don't know how you guys stay so up to date with everything, but I will definitely be back with some other questions in the future, I'm sure.  I have a lot to learn.  You guys ended my weeks of frustration.  Thank you!

Wmelonhead, 2022-08-28:
I will close the ticket now....if I can figure out how to do that.  Lol

closed 2022-08-28 by Wmelonhead

radek8, 2022-08-30:
@Wmelonhead 
Back button fix has been merged