bigtreetech/BIGTREETECH-TouchScreenFirmware #2592: [BUG] (BTT UPS and plr_z_raise)

LinoFG, 2022-08-30:
### Description
In release 27 it worked, now with the power outage recovery, the printer restarts at the wrong height + 10mm, ingnoring parameter pl_recovery_z_raise.
when trigger power outage, the bed lowered of 10mm as predicted by the marlin firmware.
After the resumption of printing, the bed is lowered further by 10mm.
After heating the bed and the nozzle, the bed rises only 10mm and consequently 10mm are uncovered.

### Steps to reproduce
Enable power loss recovery.
Disable to reduce the loss of TFT SD card or TFT USB disk.
Options: [disable: 0, enable: 1]
pl_recovery:1

Power Loss Recovery Homing
Home before power loss recovery.
Options: [disable: 0, enable: 1]
pl_recovery_home:0

Power Loss Recovery Z Raise
Raise Z axis on resume (on power loss with UPS).
Unit: [distance in mm]
Value range: [min: 0.0, max: 2000.0]
pl_recovery_z_raise:10.0

BTT Mini UPS Support
Enable backup power/UPS to move Z axis on power loss.
Options: [disable: 0, enable: 1]
btt_mini_ups:1

kisslorand, 2022-08-30:
> when trigger power outage, the bed lowered of 10mm as predicted by the marlin firmware.
> After the resumption of printing, the bed is lowered further by 10mm.

Maybe this is the problem. The TFT doesn't know Marlin raised the head or lowered the bed. What if you disable this function in Marlin?

LinoFG, 2022-08-31:
I could try, I don't understand why to do it, I lose the functionality of the ups module. The module
"BIGTREETECH-MINI-UPS" is designed to raise the nozzle when the power is cut. The "btt_mini_ups: 1" and "pl_recovery_z_raise: 10.0" parameters should indicate that the UPS module is present and the nozzle rises 10mm when fired trigger power outage. in release 27 it works like this.

closed 2022-09-14 by bigtreetech