bigtreetech/BIGTREETECH-TouchScreenFirmware #2594: Why can't I install new firmware to BTT TFT50?

AbyssNullImage, 2022-09-01:
I had installed the newer version of the firmware prior, following the instructions outlined here on this page.
~ BIGTREE_TFT50_V3.0.27.X ~
Was working fine, but noticed some settings or functions were not enabled.
Looked up a video as to how to modify and what things to modify.
Informed by video to change things in VScode and to make changes in the configuration.h file in the src.
Made changes, compiled and then proceed to try two methods, the original method with all three files.
When it failed to update, ie knob not changing colour or brightness remaining the same.
Then tried without the config.ini, to see if anything would change. Still nothing.
Although, it would update the logos/pictures, minus the PID since it could not locate, but I don't believe that is enough to cause it not to go through. It stated the same thing when I first updated it.
It would change the config.ini to CUR and the folder containing the photos into CUR, but not the BIN file.
Went through the process of both soft resetting and hard resetting, where a reset.txt was placed in the sd card.
Screen did its thing, calibrate and spit out a reset.txt.DONE
Tried to flash new firmware once more, and still nothing.
[image]
[image]
[image]

Can anyone tell me where I'm going wrong specifically? Am I missing something obvious?

radek8, 2022-09-01:
What version of the display do you have? 
ST or GD?

AbyssNullImage, 2022-09-01:
I can't recall, when I got it through Amazon, it didn't state anything like that.
I did notice there was an option for the TFT70 but it did have a (GD) attached to it, but not the TFT50 I ordered.

radek8, 2022-09-01:
The link is to versions V3.0.1
That should be the GD version. Check the processor on the TFT.
If the processor model is GD, you must use FW version GD
So you used the wrong FW version. That's why you have a problem..

AbyssNullImage, 2022-09-01:
I will take my screen out later on and check out the processor.
If I notice the model number or letter starts with GD, I will correct.
If so, I will keep you updated. I didn't see GD in the page, so I thought it wouldn't be that.
Again, thank you  for assisting me. I hope to come back with some good news.

AbyssNullImage, 2022-09-02:
It worked, but still nothing has really changed.
I did check the processor, and it does start with GD.
Default for this I guess is knob off.
So followed the process of trying to change some of the config following the instructions outlined on the page.
Compiled, and looks like it worked too. It changed all the files into CUR too.
Although, the change don't seem to take effect.
[image]
Thank you for all the assistance, but I'm not sure what do next.
TL: DR
So original question was answered, and was able to update. Processor on board started with GD.
Needed GD version as Radek8 stated. Put on SD card, seemed to update.
But doing custom config did not take effect.

closed 2022-09-02 by AbyssNullImage

kisslorand, 2022-09-02:
Did you updated with the newly compiled FW or the one from the "Copy to SD Card root directory to update"? The one from this folder is precompiled with the default settings, the one that you compiled is in the "pio" folder.

AbyssNullImage, 2022-09-02:
I updated first with the "Copy to SD Card..." which seemed to work to first see if it was that.
Put all three files and it changed them into CUR.
I didn't see any immediate changes, since it was default.
So since it successfully changed those, I went ahead and compiled my own through VScode via PlatformIO, as outlined on the page.
Changed simple things such as knob colour and brightness to see if changes took effect.
Took the GD bin from .pio, put in the config.ini from "Copy to SD..." and lastly the Unified theme onto the card, did the whole update sequence, but nothing changed after.
Knob did not glow, and brightness was still 100%, not dimmed to 50% like stated on modded configuration.
Tried using maybe the config.ini from inside SRC to see if it would have any effect, still nothing.

radek8, 2022-09-02:
The config.ini file is a configuration file, and is used to configure TFT without having to compile your own bin souor.
You have to manually modify this ini file and upload it to TFT.
The ini file will overwrite the default settings you set in the bin file.

radek8, 2022-09-02:
There are several parameters in the BIN file that cannot be set by the INI file. But most users use the default values for these parameters and do not need to compile their own BIN file

AbyssNullImage, 2022-09-03:
I'm just following the method outlined on the page.
Again, only thing I changed was the knob colour and brightness.
Changes did not take effect. Tried without .ini file.
For all attempts, the files were turned into .CUR, but none of those times did the changes take effect.
Knob remained off, and brightness remained the same.

radek8, 2022-09-03:
The color and brightness of the button can also be set from the display menu