bigtreetech/BIGTREETECH-TouchScreenFirmware #2595: "no printer attached" Issue after Power Supply Failed / SKR Mini V2.0 / TFT35 E3 V.30 / Ender 3 /

damachine17, 2022-09-01:
My Power supply recently went out, so I got a new one on my ender 3 pro.  After installing the PS, my SKR Mini V2.0 & TFT35 V2.0  have the "no printer attached error".  Before the PS went out, everything had been working fine for almost a year.  I've doubled checked all my connections, verified the RS232 cable and pin outs are correct(they haven't changed), and I even ordered another SKR Mini V2.0 board, thinking maybe the PS fried something.  R

Running in Marlin mode prints and everything moves just fine - but switching to the BTT touchscreen mode, it doesn't work.

I've tried reflashing the latest versions for both the SKR Mini & the TFT35 Display.  Oddly, I've noticed trying to update the firmware on the display doesn't seem to be working.  Per the attached picture, this is the folder structure I use for trying to flash the TFT35 display.
[image]

This issue is very similar to [Issue #637](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/637)

Should I try a new TFT35 board?  Any thoughts as to why this all of a sudden is giving me this "no printer attached error"?  I'm at my whits end and need help!

radek8, 2022-09-01:
You have 2 problems
1. you can't update FW in TFT
2. your TFT will not connect to the SKR

Solve the first problem first, then deal with the second.
Check if you used the correct connectors (send photos)
Check the SD card. Must not be larger than 8GB.
Format it again to FAT32
Check if you have a GD or ST version TFT display. Use the correct bin file accordingly

damachine17, 2022-09-01:
I just realized I wasn't putting the config.ini onto the SD card when I was trying to flash the display.
How do I know if I have the GD or ST version of the TFT display?  When I originally installed everything and it worked, I used the BIGTREE_TFT35_V3.0_E3.26.x.bin file(now at version 27 per github).
I formatted the SD card and it's the tiny 128MB one that came with it.

The connections haven't changed since it was first working but here are pictures of the display and the SKR Mini
[image]
[image]
[image]

radek8, 2022-09-02:
You have ST versions.
!!! But you have a badly connected serial cable!!!

radek8, 2022-09-02:
You must not connect the Serial cable to the SWD connector!

damachine17, 2022-09-02:
I'm confused, my serial cable is connected to the RS232 connector and not the SWD connector on the SKR Mini...what am I missing?  And why is it a 'badly connected serial cable'?  Can you provide a picture or explanation of what is wrong with the connection?
[image]

So for the Touchscreenfirmware, which version should I be installing? 1 or 2 below?
1. BIGTREE_TFT35_V3.0.27.x.bin
2. BIGTREE_TFT35_V3.0_E3.27.x.bin

I actually tried #2 today, and it flashed onto the display just fine, but I still got a "no printer attached" afterward.  Turns out my SD card slot has a bad connection, so I had to physically put pressure on the backside of the metal housing for it to read the SD card.  In either case the firmware loaded fine.
[image]

Additionally, I didn't know this, but you can ONLY flash the firmware with an SD card and not a USB SD card reader(which I had been using in the past).

damachine17, 2022-09-02:
Are you referring to the black cable going into SWD connector on the SKR Mini?
[image]

radek8, 2022-09-02:
> Are you referring to the black cable going into SWD connector on the SKR Mini?

Yes

damachine17, 2022-09-02:
Omg...I just realized I had the black cable plugged into the wrong ports on the SKR Mini!  I don't know how I missed this, but great catch!  Once I put the cable in the right place it was working again.  

Just to clarify though, which version of firmware should I be using for the display?
1. BIGTREE_TFT35_V3.0.27.x.bin
2. BIGTREE_TFT35_V3.0_E3.27.x.bin

Thank you so much for your help!!!

radek8, 2022-09-02:
> I'm confused, my serial cable is connected to the RS232 connector and not the SWD connector on the SKR Mini...what am I missing? And why is it a 'badly connected serial cable'? Can you provide a picture or explanation of what is wrong 
> 
> So for the Touchscreenfirmware, which version should I be installing? 1 or 2 below?
> 
> 1. BIGTREE_TFT35_V3.0.27.x.bin
> 2. BIGTREE_TFT35_V3.0_E3.27.x.bin
> 
> I actually tried #2 today, and it flashed onto the display just fine, but I still got a "no printer attached" 

Yes #2 is correct.

damachine17, 2022-09-02:
Thank you so much for your help...you're awesome!

radek8, 2022-09-02:
That's great. You can close the ticket.

damachine17, 2022-09-02:
Had black cable plugged into wrong port the entire time - double check connections for future reference to all who read.

Thanks again @radek8 for your help!

closed 2022-09-02 by damachine17