bigtreetech/BIGTREETECH-TouchScreenFirmware #2598: G-code thumbnails

wigw, 2022-09-05:
Hello group.
My hardware is BIGTREE_TFT70_V3_0.
I am trying to enable inline G-code thumbnails #define THUMBNAIL_PARSER 2 then compile.
For some reason, thumbnails are not displayed on TFT, but in OctoPi they are displayed.
Maybe I'm doing it wrong.
Please point me to the right path.**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

sarvenn, 2022-09-12:
I'm still using this tool with latest PrusaSlicer without any issues.
I couldn't use thumbnail parser as well.
Even prusaslicer or superslicer has also added different thumbnail options as I had a previous FR for it, but I still like to use the standalone python script.
I'm using pre-compiled TFT firmware for my TFT35

wigw, 2022-09-14:
Many thanks to everyone, your advice helped me a lot.

oldman4U, 2022-09-17:
Please help the community and close the ticket once you do not need it anymore. Thank you

wigw, 2022-09-17:
Huge thanks to everyone who helped

closed 2022-09-17 by wigw